arduino/Arduino #7867: [Board manager] 3rd party manager link for RobotCing

StanislavJochman, 2018-08-07:
I would like to add the following board manager json link to the Unofficial list of 3rd party boards support urls.

This board manager is used in Robot Cing project.(https://github.com/RobotCing/Cing/blob/master/README-en.md)

link for board manager json file(https://raw.githubusercontent.com/RobotCing/Cing/master/Software/Packages/index/package_RobotCing_index.json)

Best regards
Stanley
member of RobotCing Team

per1234, 2018-08-07:
I have added it:
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls/_compare/4d939f0a6a507f6712f5a3b8149c19276a4b0094...630989d458f2d8bf4f7fea50cef08801b852e637

For future reference, the wiki pages in this repository are editable by anyone. So you are welcome to make any improvements you like to them directly, rather than needing to open an issue report.

closed 2018-08-07 by per1234

StanislavJochman, 2018-08-07:
I just wasn't sure.

Thanks a lot