arduino/Arduino #7870: shiftIn and shiftOut do not work with SPI_MODE 3 and 4

AndySymons, 2018-08-09:
The solution described here might well fix several other problems in this thread where bits go missing or additional bits appear...  
I was trying to use the standard Arduino shiftIn and shiftOut to operate a 74HC165 shift register. I found that it always clocked one too many bits, and therefore shifted all the bits to the left, losing the top one. This did not happen when I used hardware SPI (the SPI library). 
I therefore took a close look at the source code for shiftIn and shiftOut in (wiring_shift.c) and soon spotted the problem: shiftIn and shiftOut assume a clock polarity of 0. _**They will therefore only work with SPI_MODE0 and 1.**_ 

This can be fixed by inverting the clock signal if the SPI_MODE is 3 or 4. I give my solution below. The functions require the SPI mode as an additional parameter, using the Arduino standard definitions SPI_MODE0 etc. 
While I was at it, I gained a little speed by taking the test for  MSBFIRST or LSBFIRST outside the bit loop and by declaring the method ‘inline’. 
I have tested NEWshiftIn with the 74HC165 shift register (SPI_MODE3) and it works fine. I have not tested NEWshiftOut yet as I do not have a MOD3 or 4 device output to test it with. I have also not tested the timings to see if there is a real improvement, as I do not have access to the necessary equipment at the moment. I will be happy to hear from anyone who does any of these tests.  
Happy shifting in SPI_MODES 3 and 4! 

// ===============================================================================================================================
// New version of shiftIn() and shiftOut with the clock polarity bug removed
// Requires SPI_MODE as a parameter
// Also speeded up by taking the bit direction test out of the loop
// Andrew W Symons 9-Aug-2018
//
// SHIFT IN
//
inline uint8_t NEWshiftIn(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t spiMode )
{
    uint8_t value = 0;
    uint8_t i;
    
    uint8_t clock_polarity = spiMode >> 1 ;
    switch ( bitOrder )
    {
        case LSBFIRST:
        {
            for (i = 0; i < 8; ++i)
            {
                digitalWrite(clockPin, HIGH ^ clock_polarity );
                value |= digitalRead(dataPin) << i;
                digitalWrite(clockPin, LOW ^ clock_polarity );
            } ;
        } ;
        break ;
    
        case MSBFIRST:
        {
            for (i = 0; i < 8; ++i)
            {
                digitalWrite(clockPin, HIGH ^ clock_polarity );
                value |= digitalRead(dataPin) << (7 - i);
                digitalWrite(clockPin, LOW ^ clock_polarity );
            } ;
        } ;
        break ;
    } ;
    return value;
} ;
//
// ===============================================================================================================================
// SHIFT OUT
//
inline void NEWshiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val, uint8_t spiMode )
{
    uint8_t i;
    uint8_t clock_polarity = spiMode >> 1 ;
    switch ( bitOrder )
    {
        case LSBFIRST:
        {
            digitalWrite(dataPin, !!(val & (1 << i)));
            digitalWrite(clockPin, HIGH ^ clock_polarity );
            digitalWrite(clockPin, LOW ^ clock_polarity );
        } ;
        break ;
    
        case MSBFIRST:
        {
            digitalWrite(dataPin, !!(val & (1 << (7 - i))));
            digitalWrite(clockPin, HIGH ^ clock_polarity );
            digitalWrite(clockPin, LOW ^ clock_polarity );
        } ;
        break ;
    } ;
} ;
//
// ===============================================================================================================================
// END OF FILE
//

sandeepmistry, 2019-09-16:
Hi @AndySymons,

What board do you see this behaviour on?

AndySymons, 2019-09-17:
It was an Arduino Micro clone on a breadboard wired to a 74HC165 shift register. Is this relevant? The error in the code is apparent by inspection. The core code should be corrected in all versions to work correctly with  SPI_MODE 3 and 4. My solution is, of course only an example of how it could be done.

cmaglie, 2019-09-17:
The reference page https://www.arduino.cc/reference/en/language/functions/advanced-io/shiftout/ proposes:

```if you’re interfacing with a device that’s clocked by rising edges, you’ll need to make sure that the clock pin is low before the call to shiftOut(), e.g. with a call to digitalWrite(clockPin, LOW).```

(also the accompanying example uses an 74HC565 that, I suppose, has inverse logic for clock pin compared to 74HC165, so SPI mode 0 or 1).

Anyway the suggested change is not backwards compatible, if implemented it will break all the existing sketches (causing a syntax error for the missing parameter).

AndySymons, 2019-09-17:
Yes, that is all correct. I guess the old interface would have to be left for compatibility, but clearly documented is only for mode 0 or 1, and marked as obsolete. A new interface should be added that copes with all four modes.