arduino/Arduino #7871: Please add the GenericMotorDriver library

NicholasBerryman, 2018-08-09:
Please add the following library to the library manager:

https://github.com/NicholasBerryman/GenericMotorDriver

Thanks.

per1234, 2018-08-09:
Each library must have a unique library.properties `name` value. The `name` value you used in your library.properties: `USBControllerLib` is already taken (by your other library of that name). Please change your library.properties to have a unique name value and then create a new tag/release or redo the 0.6.0 tag.

Once that's done, please comment here to let us know we can proceed.

NicholasBerryman, 2018-08-09:
Sorry about the mistake, it's all fixed now.

Thanks.

per1234, 2018-08-10:
The Library Manager indexer works off of Git tags (or GitHub releases). Although the name has been changed, the repository still had the incorrect name at the time your 0.6.0 tag was created:
https://github.com/NicholasBerryman/GenericMotorDriver/blob/0.6.0/library.properties#L1

Please either create a new tag/release (after updating the `version` value in library.properties) or else redo the 0.6.0 tag.

NicholasBerryman, 2018-08-11:
I've edited the release to include the updated library.properties,

Thanks

cmaglie, 2018-08-17:
Done!

closed 2018-08-17 by cmaglie