arduino/Arduino #7880: Serial Connection missing from port selection - MacOS Mojave Build 18A353d

joshrawls, 2018-08-12:
’m not sure what happened as this was working without issue previously. However, the serial port that used to show up in the Arduino IDE no longer shows. This makes it impossible to upload programs to the Arduino. I have attached a screenshot showing the RS232 device in System Information and the missing connection in the Arduino menu dropdown. What should show is something like “serial00000001”. I can’t remember exactly.

I have also sent this in to Apple.
[image]

cmaglie, 2018-08-13:
Did you upgrade MacOS or the FTDI driver?
Could you run the following command in a terminal:

```
ls /dev/
```

and see if you can find the serial port in the list? I'm interested in the exact name
(maybe try to list /dev/ with and without the board connected)

joshrawls, 2018-08-15:
Figured out the issue. Parallels Tools was the culprit. Uninstalled that and it's working again.

closed 2018-08-15 by joshrawls

cmaglie, 2018-08-17:
Thanks for the feedback with the solution. Glad you solved it!

mateothegreat, 2018-09-28:
After upgrading to Mojave as well .. I had to connect to my device directly and eliminate my usb hub out of the equation to get it to work. fyi

cskonopka, 2019-02-04:
I'm still having this issue with various microcontrollers and SBCs. I've tried trying the USB-C dongle and via hub with no luck. I've also tried several drivers as well, definitely getting frustrating at this point. Anyone find a solid solution for Mojave?

joshrawls, 2019-02-04:
Going through a USB3 hub seemed to do the trick for me.

Josh Rawls
joshua.rawls@icloud.com