arduino/Arduino #7884: Change repository for library

Manghao, 2018-08-13:
Hello, I've new github account for professional life, I developed 2 libraries ([RBGLed](https://github.com/Manghao/RGBLed) and [LM35](https://github.com/Manghao/LM35)), I want transfer this libraries in my new account, How can I do it to be updated for arduino ?

DrDiodac, 2018-08-13:
Hi. They have tutorial about how to do. But if you use IDE maybe is options for connection your IDE with your account and you can use GitHub from IDE for downloading, search and making repository. Some IDE allows it.

Manghao, 2018-08-13:
Where can I find this tutorial?

per1234, 2018-08-13:
There is no tutorial. This has absolutely nothing to do with the Arduino IDE. It's about the indexer job that generates the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) from the Git tags in a list of repositories.

If you transfer ownership of the repositories to your new account then GitHub will automatically redirect from the old URL and I believe the Library Manager indexer job will follow that redirect, so it should not be necessary to do anything.

https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/

If you need to make the transfer in a way that doesn't result in a redirect then someone from Arduino will need to update the URLs for your libraries to the new ones.

per1234, 2018-08-13:
@DrDiodac that tutorial has no relevance to this conversation and Manghao obviously already knows all that information. Please don't comment here unless you have some information of value to add.

Manghao, 2018-08-14:
@per1234 I made transfers into my new account but there is no change of deposit here http://downloads.arduino.cc/libraries/library_index.json.

I have to make a new version or a new request?

per1234, 2018-08-14:
It's not clear what you mean by "change of deposit".

The Library Manager indexer job is going to continue to go to the repository URL added after your original request to add your libraries. But since you did a proper repository transfer, GitHub automatically redirects that URL to your new repository URLs so that doesn't matter. Your existing releases will continue to be available for installation via Library Manager and any future releases you make will also be picked up by the indexer job. So I don't see any problem.

Manghao, 2018-08-14:
@per1234 So if I understand correctly everything is good.
Thank you for your help.

cmaglie, 2018-08-17:
Since we are here, even if not strictly needed, I've changed the URL in the library manager.

closed 2018-08-17 by cmaglie