arduino/Arduino #7887: [Library Manager] Please add libraries

hideakitai, 2018-08-15:
Hello, I created some libraries. 
Could you add following to a list of libraries ?

DRV2667 Piezo Haptic Driver
https://github.com/hideakitai/DRV2667

MAX17048/MAX17049 1-Cell/2-Cell Fuel Gauge
https://github.com/hideakitai/MAX17048

MCP4728 quad channel, 12-bit voltage output DAC
https://github.com/hideakitai/MCP4728

MPU9250 Nine-Axis (Gyro + Accelerometer + Compass)
https://github.com/hideakitai/MPU9250

PCA9536 4-Bit I2C and SMBus I/O Expander
https://github.com/hideakitai/PCA9536

PCA9547 (8-channel I2C-bus multiplexer with reset)
https://github.com/hideakitai/PCA9547

RTC PCF2129 with integrated quartz crystal
https://github.com/hideakitai/PCF2129

Benewake TFmini micro LiDAR module
https://github.com/hideakitai/TFmini

Midi Time Code (MTC) parser
https://github.com/hideakitai/MTCParser

binary data packer / unpacker
https://github.com/hideakitai/Packetizer

Thanks.

per1234, 2018-08-15:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) in each repository that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hideakitai, 2018-08-16:
Thanks, I released each libraries.

cmaglie, 2018-08-17:
There are already libraries named PCA9536 and TFMini, you must rename them:
1) change the name in library.properties
2) commit and push
3) delete and redo the release tag.

The other instead have been added correctly.

hideakitai, 2018-08-19:
Thanks, `PCA9536` was renamed to `PCA9536Arduino` and updated to `0.1.1`.
I think `TFmini` is unique if you treat upper/lower cases as different character. How about it?

[image]

cmaglie, 2018-08-22:
> I think TFmini is unique if you treat upper/lower cases as different character. How about it?

No, please, change the name, there are file systems that are not case sensitive, better to avoid this kind of problems from the beginning.

hideakitai, 2018-08-23:
OK, I have changed the name of library.properties and released 0.1.1.
Please check it out.
Thank you for your cooperation.

per1234, 2018-08-23:
Due to the name change, I'm not sure whether it will be an issue this time or not, but please make sure to always update the `version` field in library.properties before you make a release. Otherwise the release will not be recognized by the Arduino Library Manager.

hideakitai, 2018-08-23:
Sorry, I also changed the version of library.properties and released 0.1.1 again. Check it out please.

cmaglie, 2018-08-24:
All libraries have been added.

closed 2018-08-24 by cmaglie