arduino/Arduino #7895: MKRZERO USB problems under OS X

Lilyjan, 2018-08-16:
When uploading a sketch from Arduino IDE under OS X to a MKRZERO, as soon as the upload completes, the serial port becomes deselected. The serial port can be temporarily reconnected with a ‘double-click’ reset on the board. However, the sketch will not execute and any attempt to launch the Serial Monitor also disconnects the port. This is a documented problem on the Web, but the only suggested solution I have found is to install ch34x drivers (e.g. from GitHub). However, in my case this has made no difference. This problem seems reproducible on any MKRZERO and under OS X 11, 12 and 13. Isn't there a robust solution to this?

[image]

sandeepmistry, 2018-08-16:
Could you please provide the exact sketch you are uploading.

Does it happen with all sketches or just one?

Lilyjan, 2018-08-17:
Thanks so much for this! It never occurred to me that a sketch that compiles correctly might cause such issues!! I now know that I do indeed have sketches that do not exhibit this error. 
All my sketches which cause deselection of the USB port turned out to be (larger) variants of the one I have uploaded. By trial and error, I have narrowed the problem down to the highlighted instruction of the AudioFrequencyMeter library [ meter.begin() ]. The problem does not occur when this line is commented out. My code is closely based on the example in the Arduino documentation ‘A Simple Frequency Meter for Arduino Zero’: arduino.cc/en/Tutorial/SimpleAudioFrequencyMeter. This published example uploads and executes on my same board successfully.
Can you PLEASE help me understand why the identical command in my sketch is causing such problems?
[problemSketch.txt](https://github.com/arduino/Arduino/files/2297641/problemSketch.txt)