arduino/Arduino #7902: Core cache directory name for esp8266 is too long on Windows.

storzinc, 2018-08-19:
On Windows maximum file name length is 255 characters, but in the buld log I see this line:

`Archiving built core (caching) in: C:\Users\storzinc\AppData\Local\Temp\arduino_cache_875001\core\core_esp8266_esp8266_generic_CpuFrequency_80,VTable_flash,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_512K0,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_256000_1ee66ed3f9bdc333f8cbfdd139bc1dd8.a
`

Arduino here is trying to create a directory with 262 characters long name, as a result no cache is created and compiller is doing full rebuild every time.

Tried this on Arduino 1.8.5, Arduino 1.9-BUILD-78 and the hourly build.

facchinm, 2018-08-20:
Hi @storzinc ,
unfortunately any way to reduce that filename length is going to affect the actual functionality (let's suppose we cut the string at some point, the cached core filename could incorrectly match another core; the wrong binary would be produced, not matching the chosen preferences). 
Not using the cache in this case is slower but safer, so I'd close the issue as wontfix.

matthijskooijman, 2018-08-20:
@facchinm, there are other ways to implement matching the cache to the core. A file *inside* the cache directory containing these options would work. You would of course have to scan all cache directories to find the appropriate one, but that's not too much of a problem. An index file, which contains the mapping could also work and require only reading a single file.

facchinm, 2018-08-20:
@matthijskooijman I agree, there are a lot of alternative methods (hashing the filename with sha256 and using the hash as filename should be safe enough too :wink: ), I'd reopen the issue on `arduino-builder` as an improvement, what do you think?

matthijskooijman, 2018-08-20:
Sounds good to me.

closed 2018-09-04 by facchinm