arduino/Arduino #7903: [Library Manager] Please add libraries

hideakitai, 2018-08-20:
Hello, I created some libraries. Could you add following to a list of libraries ?

EmbeddedUtils
https://github.com/hideakitai/EmbeddedUtils

EmbeddedDevices
https://github.com/hideakitai/EmbeddedDevices

Thanks.

per1234, 2018-08-20:
The Arduino Library Manager indexer [does not work with the submodules](https://github.com/arduino/Arduino/issues/6385#issuecomment-310406324) used in your EmbeddedDevices repository. The files from the submodules will be missing. You might consider using Git [subtrees ](https://www.atlassian.com/blog/git/alternatives-to-git-submodule-git-subtree)instead of submodules. Even without the use of submodules, that repository does not have a valid [1.5 or 1.0 library structure](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#layout-of-folders-and-files). A file with a recognized header extension (.h, .hh, .hpp) must be present in either the root of the repository or in the `src` subfolder.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After resolving the problems, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hideakitai, 2018-08-20:
Ok, then could you add EmbeddedUtils only?
Sorry, and I released EmbeddedUtils with tag 0.1.0.
Thank you.

cmaglie, 2018-08-24:
Done

closed 2018-08-24 by cmaglie