arduino/Arduino #7908: Web interface programmed arduino wont boot without computer.

HarveyBrezinaConniffe, 2018-08-21:
I programmed an ardunio through the web interface to be used in a situation without a computer. When I booted it through USB it worked fine. However, When using a power supply or a computer without the IDE installed, the program did not boot. After hours of troubleshooting we finally tried re-flashing the program with the traditional IDE, After that it worked flawlessly and booted perfectly on an external power supply. Is this a bug in the web interface?

facchinm, 2018-08-22:
HI @HarveyBrezinaConniffe ,
we are missing some information here :wink: 
* which board are you targeting?
* which IDE version are you running?
* can you paste the complete verbose output for both compilation?
* can you provide a sketch to demonstrate the issue and a description of  what it should do?

Thanks!

HarveyBrezinaConniffe, 2018-08-22:
The board I was using was a Genuino 101. The IDE that caused the board to boot only when connected to a computer with the IDE installed was the new web IDE and the IDE that made the board work as expected was the Mac OS native IDE V1.8.5. I don't really think this was to do with my program as the exact same thing happened even when I tried to run the Blink example, But here is the output and code anyway:
Compilation output on web IDE( https://create.arduino.cc/editor ), Board does not boot unless powered by a laptop:
```
./opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -hardware opt/arduino-builder/hardware -hardware ./opt/cores -tools opt/arduino-builder/tools -tools ./opt/tools -built-in-libraries opt/libraries/latest -logger humantags -fqbn Intel:arc32:arduino_101 -build-cache /tmp -build-path /tmp/104367961/build -verbose -libraries /tmp/104367961/custom -libraries /tmp/104367961/pinned /tmp/104367961/sketch_aug21a

Sketch uses 52480 bytes (33%) of program storage space. Maximum is 155648 bytes.
```
Compilation output on the 1.8.5 Mac OS IDE, Board behaves as expected:
```
Sketch uses 25064 bytes (16%) of program storage space. Maximum is 155648 bytes.
```
Code:
``` C
#include <Key.h>
#include <Keypad.h>

//Code Lock program. Written by Harvey Brezina Conniffe.

// Initialize the keypad.
const byte ROWS = 4;
const byte COLS = 4;
char keys[ROWS][COLS] = {
  {'1','2','3', 'A'},
  {'4','5','6', 'B'},
  {'7','8','9', 'C'},
  {'#','0','*', 'D'}
};
byte rowPins[ROWS] = { 0, 1, 2, 3 };
byte colPins[COLS] = { 4, 5, 6, 7 }; 

Keypad kpd = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

// Define pins for LEDs, Piezo Buzzer and the door lock relay. 

#define red1 13
#define red2 12
#define green1 11
#define green2 10
#define piezo 9
#define unlockp 8

//The correct code to unlock the door.
String correct = "000000";
String code;
// Count of how many failed attempts.
int icount = 0;
// Happens when the code is entered wrong.
void wrong() {
  digitalWrite(green1, LOW);
  digitalWrite(red1, HIGH);
  digitalWrite(red2, HIGH);
  tone(piezo, 3000, 200);
  delay(200);
  tone(piezo, 3000, 200);
  delay(200);
  digitalWrite(red1, LOW);
  digitalWrite(red2, LOW);
  digitalWrite(green1, HIGH);
}
// Makes a little beep when a button is pressed.
void pip() {
  digitalWrite(green1, HIGH);
  tone(piezo, 4000, 200);
  delay(200);
  digitalWrite(green1, LOW);
}
// Unlocks the door and beeps.
void unlock() {
  digitalWrite(green1, HIGH);
  digitalWrite(green2, HIGH);
  digitalWrite(unlockp, HIGH);
  tone(piezo, 3000, 4000);
  delay(4000);
  digitalWrite(unlockp, LOW);
  digitalWrite(green1, LOW);
  digitalWrite(green2, HIGH);
}
// Locks the keypad for 10s after 3 failed attempts.
void fail() {
  for(int i = 0; i < 10;i++) {
    digitalWrite(red2, LOW);
    digitalWrite(green1, LOW);
    digitalWrite(red1, HIGH);
    digitalWrite(green2, HIGH);
    tone(piezo, 4000, 333);
    delay(500);
    digitalWrite(red1, LOW);
    digitalWrite(green2, LOW);
    digitalWrite(red2, HIGH);
    digitalWrite(green1, HIGH);
    tone(piezo, 2000, 333);
    delay(500);
  }
  digitalWrite(red2, LOW);
  digitalWrite(green1, LOW);
  digitalWrite(green2, HIGH);

}
void setup() {
    pinMode(red1, OUTPUT);
    pinMode(red2, OUTPUT);
    pinMode(green1, OUTPUT);
    pinMode(green2, OUTPUT);
    pinMode(piezo, OUTPUT);
    pinMode(unlockp, OUTPUT);
    digitalWrite(red1, LOW);
    digitalWrite(red2, LOW);
    digitalWrite(green1, LOW);
    digitalWrite(green2, HIGH);
    digitalWrite(piezo, LOW);
    digitalWrite(unlockp, LOW);
}
void reset() {
  icount = 0;
  code = "";
}
void loop() {
  char key = kpd.getKey();
  if(key) {
    pip();
    code += key;
    if(code == correct) {
      unlock();
      reset();
    }
    else if(code.length() == correct.length()) {
      wrong();
      icount ++;
      if(icount == 3) {
        fail();
        reset();
      }
      code = "";
    }
  }
}
```
Thanks!

facchinm, 2018-08-23:
Got it, thanks!
The difference seem to be about the core version in use; can you confirm me that you are using version 2.0.2 of the Curie core? I'll investigate with the web team to understand what they're using on their side.
Thank you for all the info!

e-ika, 2018-09-06:
This seems interesting for me

per1234, 2018-09-26:
@HarveyBrezinaConniffe did you get a chance to update to Intel Curie Boards 2.0.2? If so, did it resolve your issue?