arduino/Arduino #7914: [Library Manager] Library add request

maisonsmd, 2018-08-23:
Requesting the addition of my library https://github.com/maisonsmd/msTask to the library manager.
This lightweight library emulates multitasking in Arduino

per1234, 2018-08-23:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

maisonsmd, 2018-08-23:
I updated them, please review!
Thank you!

cmaglie, 2018-08-24:
Done

closed 2018-08-24 by cmaglie