arduino/Arduino #7916: IDE adding comments to macros name

FernandoGarcia, 2018-08-24:
Hi!
Actually is impossible compile Marlin firmware because the IDE is adding comments to macros name or something like this.

The problem was reported by another user here: https://github.com/MarlinFirmware/Marlin/issues/11585

The errors list is on issue above.

I'm facing the same problem with 1.8.6.

Best regards

per1234, 2018-08-24:
Closing as duplicate of https://github.com/arduino/arduino-preprocessor/issues/6

closed 2018-08-24 by per1234

per1234, 2018-08-26:
It looks like I was wrong about this being arduino/arduino-preprocessor#6. See https://github.com/arduino/Arduino/issues/7930#issuecomment-416079657. Since arduino/Arduino#7930 has more information, I'll leave this closed.