arduino/Arduino #7923: arduino 1.8.6 avrdude error on raspberry pi

xiiSharky, 2018-08-25:
I've been trying to get this working for awhile now but whatever I do, it fails due to the same error. Ive tried completely reinstalling my rasbian distro and my arduino IDE but whatever I do, I always get this error: 
Arduino: 1.8.6 Hourly Build 2018/08/23 11:19 (Linux), Board: "Arduino/Genuino Uno"

/home/pi/Downloads/arduino-nightly/arduino-builder -dump-prefs -logger=machine -hardware /home/pi/Downloads/arduino-nightly/hardware -tools /home/pi/Downloads/arduino-nightly/tools-builder -tools /home/pi/Downloads/arduino-nightly/hardware/tools/avr -built-in-libraries /home/pi/Downloads/arduino-nightly/libraries -libraries /home/pi/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10806 -build-path /tmp/arduino_build_455901 -warnings=none -build-cache /tmp/arduino_cache_441874 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/pi/Downloads/arduino-nightly/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/pi/Downloads/arduino-nightly/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/pi/Downloads/arduino-nightly/hardware/tools/avr -verbose /tmp/arduino_modified_sketch_655932/Blink.ino
/home/pi/Downloads/arduino-nightly/arduino-builder -compile -logger=machine -hardware /home/pi/Downloads/arduino-nightly/hardware -tools /home/pi/Downloads/arduino-nightly/tools-builder -tools /home/pi/Downloads/arduino-nightly/hardware/tools/avr -built-in-libraries /home/pi/Downloads/arduino-nightly/libraries -libraries /home/pi/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10806 -build-path /tmp/arduino_build_455901 -warnings=none -build-cache /tmp/arduino_cache_441874 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/pi/Downloads/arduino-nightly/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/pi/Downloads/arduino-nightly/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/pi/Downloads/arduino-nightly/hardware/tools/avr -verbose /tmp/arduino_modified_sketch_655932/Blink.ino
Using board 'uno' from platform in folder: /home/pi/Downloads/arduino-nightly/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/pi/Downloads/arduino-nightly/hardware/arduino/avr
Detecting libraries used...
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/pi/Downloads/arduino-nightly/hardware/arduino/avr/cores/arduino -I/home/pi/Downloads/arduino-nightly/hardware/arduino/avr/variants/standard /tmp/arduino_build_455901/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/pi/Downloads/arduino-nightly/hardware/arduino/avr/cores/arduino -I/home/pi/Downloads/arduino-nightly/hardware/arduino/avr/variants/standard /tmp/arduino_build_455901/sketch/Blink.ino.cpp -o /tmp/arduino_build_455901/preproc/ctags_target_for_gcc_minus_e.cpp
/home/pi/Downloads/arduino-nightly/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_455901/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/pi/Downloads/arduino-nightly/hardware/arduino/avr/cores/arduino -I/home/pi/Downloads/arduino-nightly/hardware/arduino/avr/variants/standard /tmp/arduino_build_455901/sketch/Blink.ino.cpp -o /tmp/arduino_build_455901/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /tmp/arduino_cache_441874/core/core_arduino_avr_uno_b6d012ae8ad6c0f72456763811ddfca7.a
Linking everything together...
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-gcc -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_455901/Blink.ino.elf /tmp/arduino_build_455901/sketch/Blink.ino.cpp.o /tmp/arduino_build_455901/../arduino_cache_441874/core/core_arduino_avr_uno_b6d012ae8ad6c0f72456763811ddfca7.a -L/tmp/arduino_build_455901 -lm
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_455901/Blink.ino.elf /tmp/arduino_build_455901/Blink.ino.eep
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_455901/Blink.ino.elf /tmp/arduino_build_455901/Blink.ino.hex
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avr-size -A /tmp/arduino_build_455901/Blink.ino.elf
Sketch uses 950 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
An error occurred while uploading the sketch
/home/pi/Downloads/arduino-nightly/hardware/tools/avr/bin/avrdude -C/home/pi/Downloads/arduino-nightly/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyAMA0 -b115200 -D -Uflash:w:/tmp/arduino_build_455901/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/pi/Downloads/arduino-nightly/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyAMA0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200

per1234, 2018-08-25:
Does the problem still occur if you use Arduino IDE 1.8.5?

There was a [report on the forum](http://forum.arduino.cc/index.php?topic=565163) from someone who had a similar mysterious avrdude failure with no error message but it only happened with the 6.3-20171130 version. Once they switched to a different avrdude version (still using the Arduino IDE 1.8.6 release) the problem went away.

xiiSharky, 2018-08-26:
I installed 1.8.5 and tried the same thing I was doing on the other IDE and it still didn’t work

xiiSharky, 2018-08-26:
Turns out I was using the wrong serial port the whole time, case closed

closed 2018-08-26 by xiiSharky