arduino/Arduino #7924: Lag when changing to File/Tools tab on macOS

45gfg9, 2018-08-25:
OS: macOS 10.13.6
IDE: 1.8.6
When changing between tabs, File tab causes lag. 
[image]
When changing from other tabs to File tab, it will lag about a second. 
Changing to Tools tab also causes lag, but it's much shorter, about 0.1 sec. 
Other tabs Edit/Sketch/Help tabs work fine.

MCUdude, 2018-08-26:
I'm experiencing the exact same issues. I'm using MacOS 10.12.6

PaulStoffregen, 2018-08-26:
Random question - Are you using Teensy, or have you run the Teensyduino installer on this copy of Arduino 1.8.6?  There may be a bug in Teensyduino which adds lag to some of the menus.  I'm investigating...

Just want to confirm these are the Arduino IDE *without* Teensy's modifications.  Wouldn't want to bother the Arduino devs with something that might be my fault.

MCUdude, 2018-08-26:
my menu is really laggy and slow without teensyduino installed. I ended up going back to 1.8.5 because it might take half a minute for the boards menu to respond. Instant spinning ball..

45gfg9, 2018-08-27:
Hiya, here to say I am not using Teensy.

DrDiodac, 2018-08-27:
I have the same problem huge lags, slow down even restarting itself my Mac and after restart, kernel panic alert. 
But if I not touch menu, I can working normally.

facchinm, 2018-08-27:
Bisecting right now, I'll notify you when I get something :wink:

facchinm, 2018-08-27:
Found the culprit, it's Java (again) :rage: 
1.8.5 was bundled with JRE8u151 while 1.8.6 with u171 . Bisecting shows that the first problematic version is u161 but there's nothing in the release notes that catches the eye.

The slowdown exists even if I revert https://github.com/arduino/Arduino/commit/a2b5f661b85dc7591b40cdcc5d81904e5707fe2f and companion commits or compile IDE 1.8.4, so the bug lies somewhere else.

As a workaround while we investigate, I'd revert jre to u151 on nighlty builds (even if it's EOL) since it doesn't look like there are super important bugfixes between 151 and 181.

facchinm, 2018-08-27:
@DrDiodac @45gfg9 @MCUdude would you mind testing the build that is being produced by #7935 ? Thanks!

MCUdude, 2018-08-27:
I just downloaded and tested it. Works fine for me now! 👍🏻

DrDiodac, 2018-08-27:
Yeah I can lead test,  just I need reading about changes which I need to do or send me please steps tips how to, and I can start quickly. 

Send from iPhone

45gfg9, 2018-08-28:
Thank you! Now it works as normal. No more lags. 🎉

DrDiodac, 2018-08-28:
> @DrDiodac
>
> Download http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-7935-BUILD-760-macosx.zip
> Use the downloaded test version of the Arduino IDE just as you would the production IDE release (more info)
> Thanks!

At first glance seems that slow down problems is solved! I do more testing but I see is better than before. Teensy 3.2 tested. 
Thank you!! 

Send from iPhone

Vipas-ana, 2018-08-28:
[#7935](https://github.com/arduino/Arduino/pull/7935) Fixes this issue.

45gfg9, 2018-08-28:
Ok, so now I think I can close this issue. Thank you all!

PR #7935 (2018-08-28): Fix OSX "slow" menus
JRE 8u161 made `JMenu.insert()` MUCH slower on OSX.
This exposed a bug lurking there for years; in fact the menu entries were rebuilt every time since `JMenu.getComponent()` returns an empty list.
The correct function is `JMenu.getMenuComponents()`
In the meantime, also remove `sketchbookMenu` and `examplesMenu` deletion on board change; only their content needs to be updated, not the container itself.

Fixes #7924
commit 2f3a561c (2018-08-28): Fix rebuild being triggered on every menu open()

Fixes #7924

JRE 8u161 made JMenu.insert() MUCH slower on OSX.
This exposed a bug lurking there for years; in fact the menu entries were rebuilt every time since JMenu.getComponent() returns an empty list.
The correct function is JMenu.getMenuComponents().
In the meantime, also remove sketchbookMenu and examplesMenu being deleted on board change; only their content needs to be updated, not the container itself.
closed 2018-08-28 by 45gfg9