arduino/Arduino #7930: does not give a valid preprocessing token/Since 1.8.6

Keugon, 2018-08-26:
PreWord iam not a Programmer just using Marlin for my 3d Printer
This Issue only appears when I try to compile it with 1.8.6 by 1.8.4 everything is fine i hope someone can identify the "Real Bug" here 
*****Bug report****
[Marlin Software.zip](https://github.com/arduino/Arduino/files/2321665/Marlin.Software.zip)

Arduino: 1.8.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\Marlin_main.cpp:247:

pins_GT2560_REV_A.h:93:32: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_DDR" does not give a valid preprocessing token

 #define SUICIDE_PIN        54  // Must be enabled at startup to keep power flowing

                                ^

sketch\fastio.h:95:36: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                    ^

sketch\fastio.h:113:30: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO, v) do{ SET_OUTPUT(IO); WRITE(IO, v); }while(0)

                              ^

sketch\Marlin_main.cpp:914:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(SUICIDE_PIN, HIGH);

     ^

sketch\Marlin_main.cpp:914:15: note: in expansion of macro 'SUICIDE_PIN'

     OUT_WRITE(SUICIDE_PIN, HIGH);

               ^

pins_GT2560_REV_A.h:93:32: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_PIN" does not give a valid preprocessing token

 #define SUICIDE_PIN        54  // Must be enabled at startup to keep power flowing

                                ^

sketch\fastio.h:95:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)

                                                             ^

sketch\fastio.h:113:30: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO, v) do{ SET_OUTPUT(IO); WRITE(IO, v); }while(0)

                              ^

sketch\Marlin_main.cpp:914:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(SUICIDE_PIN, HIGH);

     ^

sketch\Marlin_main.cpp:914:15: note: in expansion of macro 'SUICIDE_PIN'

     OUT_WRITE(SUICIDE_PIN, HIGH);
[... 704 lines cut ...]

sketch\fastio.h:113:46: note: in expansion of macro 'WRITE'

 #define OUT_WRITE(IO, v) do{ SET_OUTPUT(IO); WRITE(IO, v); }while(0)

                                              ^

sketch\Marlin_main.cpp:927:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(SUICIDE_PIN, LOW);

     ^

sketch\Marlin_main.cpp:927:15: note: in expansion of macro 'SUICIDE_PIN'

     OUT_WRITE(SUICIDE_PIN, LOW);

               ^

exit status 1
pasting "/* Must be enabled at startup to keep power flowing*/" and "_DDR" does not give a valid preprocessing token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

DominikPalo, 2018-08-26:
Related reports:
https://github.com/MarlinFirmware/Marlin/pull/11647
https://github.com/MarlinFirmware/Marlin/issues/10862
https://github.com/arduino/arduino-preprocessor/issues/6

per1234, 2018-08-26:
I previously thought this was arduino/arduino-preprocessor#6 but now I understand that Arduino IDE 1.8.6 doesn't actually use arduino-preprocessor (despite bundling that tool). The [MCVE ](https://github.com/arduino/arduino-preprocessor/issues/6#issuecomment-354014431)in arduino/arduino-preprocessor#6 doesn't result in an error in Arduino IDE 1.8.6. When I replace Arduino IDE's [arduino-builder 1.4.0](https://github.com/arduino/arduino-builder/releases/tag/1.4.0) with the previous [arduino-builder 1.3.25 release](https://github.com/arduino/arduino-builder/releases/tag/1.3.25) the bug reported here does not occur. So this is an arduino-builder bug rather than an arduino-preprocessor bug.

Keugon, 2018-08-27:
Huh not epected that reaction on that just wana say thank you for care about my porly sumited issue

closed 2018-09-10 by cmaglie