arduino/Arduino #7941: [Library Manager] Add MultiTrans

feklee, 2018-08-28:
Please add my library:

https://github.com/feklee/MultiTrans

_This library is an implementation of a multi channel transceiver: Each pin can transmit and receive at the same time (transceive). Multiple pins can transceive simultaneously. Collisions are automatically detected and circumvented._

matthijskooijman, 2018-08-28:
@feklee looks like an interesting library, I've been working on something similar a long time ago (but ended up with something way to complicated). One suggestion wrt terminology: You say in the README "Transmit and receive simultaneously via a single pin.". I suspect that any given time, you will be either transmitting *or* receiving, so it's not really simultaneously. IOW, the pin is half-duplex, not full-duplex (but you can use the same pin for receiving and transmitting in the same setup, so it is not simplex).

cmaglie, 2018-08-28:
Library added.

closed 2018-08-28 by cmaglie

feklee, 2018-08-28:
> I suspect that any given time, you will be either transmitting or receiving, so it's not really simultaneously.

Thing is, the library takes care of sorting out collisions (by checking the status of the line before sending a character). The user can just send. In contrast to libraries such as [SoftwareSerialWithHalfDuplex][1], there is no need to manually switch between receiving and transmitting. OTOH *MultiTrans* is not compatible with standard RS-232 serial communication. I implemented a custom protocol.

[1]: https://github.com/nickstedman/SoftwareSerialWithHalfDuplex