arduino/Arduino #7945: MacOS 10.7 (Lion) no longer works starting with Java 8u162

PaulStoffregen, 2018-08-29:
Arduino 1.8.6 will not launch on MacOS 10.7 Lion, because a newer Java JRE is used which fails to start.

Here is the Java info.
https://bugs.openjdk.java.net/browse/JDK-8201414

If a JRE 8u162 or later continues to be used, these lines Info.plist should be updated to specify 10.8 as the minimum MacOS version.

    <key>LSMinimumSystemVersion</key>
    <string>10.7</string>

This will give a message to users explaining their OSX is too old, rather than the IDE briefly appearing and then crashing.

The main website download page should probably be updated from "10.7 Lion or newer".

Or an older JRE can be used to keep compatibility with 10.7.

facchinm, 2018-08-29:
I'd drop compatibility with 10.7 to allow further security update on Java side (I believe that OSX Lion's market share is very low nowadays), but we can discuss before 1.8.7 .
@cmaglie

PaulStoffregen, 2018-08-29:
Agreed, dropping Lion support makes the most sense.

I personally use Lion here for my build machine, so I can be sure my compiled binaries work all the way back to Lion.  But this isn't about me.  It's about making the best software we can.  I have 3 macs here I use for testing, so I can manage.

Just wanted to start this issue so the issue is known.  Sooner or later Info.plist and the website will need get updated.

cmaglie, 2018-08-30:
Once Java drops supports for OSX 10.7 it's almost game-over, there is not much we can do about it. So for me is ok dropping support for Lion.

PR #7957 (2018-08-31): Java now requires MacOSX 10.8 Mountain Lion or later
Fixes #7945.

Users with 10.7 will see message, instead of having the IDE appear to start and then crash for no apparent reason.

[image]
closed 2018-08-31 by facchinm