arduino/Arduino #7946: 3rd party board not working

Menotdan, 2018-08-29:
I get an error:
Arduino: 1.8.6 (Windows Store 1.8.14.0) (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.14.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino\HardwareSerial0.cpp: In function '_GLOBAL__sub_I___vector_18':

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.14.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino\HardwareSerial0.cpp:77:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.14.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.14.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

facchinm, 2018-08-29:
Hi @Menotdan ,

which 3rd party board are you referring to? The compile target is "Arduino Uno" so maybe you forgot to change it (hence the compilation error).
If it also fails wait your 3rd party board please send a sketch that reproduces the issue and the full verbose output of compilation (you can enable it under File/Preferences)

Thanks

mikrocoder, 2018-09-05:
Hi,

sorry for waiting, I have a orignal Arduino Mega2560.

Menotdan, 2018-09-07:
it is from creation crate i think its some remake ill get apic

Menotdan, 2018-09-07:
[image]

facchinm, 2018-11-29:
Closing in favour of #7949

closed 2018-11-29 by facchinm