arduino/Arduino #7948: Could not install board java ValidatorException

MaSven, 2018-08-29:
Tried to download Boardinformation from [](https://downloads.arduino.cc/CMSIS-Atmel-1.2.0.tar.bz2)
but only get an java Exception. Here is the stacktrace.

```
 Fehler beim Herunterladen von https://downloads.arduino.cc/CMSIS-Atmel-1.2.0.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Fehler beim Herunterladen von https://downloads.arduino.cc/CMSIS-Atmel-1.2.0.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.base/java.lang.Thread.run(Thread.java:844)
Caused by: java.lang.Exception: Fehler beim Herunterladen von https://downloads.arduino.cc/CMSIS-Atmel-1.2.0.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.ssl.Alerts.getSSLException(Alerts.java:198)
	at java.base/sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1974)
	at java.base/sun.security.ssl.Handshaker.fatalSE(Handshaker.java:345)
	at java.base/sun.security.ssl.Handshaker.fatalSE(Handshaker.java:339)
	at java.base/sun.security.ssl.ClientHandshaker.checkServerCerts(ClientHandshaker.java:1968)
	at java.base/sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1777)
	at java.base/sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:264)
	at java.base/sun.security.ssl.Handshaker.processLoop(Handshaker.java:1098)
	at java.base/sun.security.ssl.Handshaker.processRecord(Handshaker.java:1026)
	at java.base/sun.security.ssl.SSLSocketImpl.processInputRecord(SSLSocketImpl.java:1137)
	at java.base/sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1074)
	at java.base/sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:973)
	at java.base/sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1402)
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1429)
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1413)
	at java.base/sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:567)
	at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:163)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:385)
	at java.base/sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:290)
	at java.base/sun.security.validator.Validator.validate(Validator.java:264)
	at java.base/sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:343)
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:226)
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:133)
	at java.base/sun.security.ssl.ClientHandshaker.checkServerCerts(ClientHandshaker.java:1947)
	... 20 more
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.provider.certpath.SunCertPathBuilder.build(SunCertPathBuilder.java:141)
	at java.base/sun.security.provider.certpath.SunCertPathBuilder.engineBuild(SunCertPathBuilder.java:126)
	at java.base/java.security.cert.CertPathBuilder.build(CertPathBuilder.java:297)
	at java.base/sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:380)
	... 26 more

```
Seems like an cacert is missing here?
Using ArduinoIDE version 1.8.5 provided by [flathub](https://flathub.org/apps/details/cc.arduino.arduinoide).
Using java version 
```
java version "10.0.2" 2018-07-17
Java(TM) SE Runtime Environment 18.3 (build 10.0.2+13)
Java HotSpot(TM) 64-Bit Server VM 18.3 (build 10.0.2+13, mixed mode)
```

MaSven, 2018-08-29:
Ok seems like the flatpak build is using openjdk i think. Dont know if flatpak uses the java version of the host or not but here are the build [files](https://github.com/flathub/cc.arduino.arduinoide)

MaSven, 2018-08-30:
Also opened this ticket for the builder of the flatpak. Since java 9 the certs have changed and some got dropped. I think to resolve this they have to build against openjdk-8 and also bundle it with this version.

MChima, 2019-05-03:
Hello,
I have the same issue with the internet from the company I'm in. On the personal laptop (with Windows 10) I do not encounter it.
Issue is happening with Arduino IDE 1.88 and 1.89, but doesn't happen with version 1.8.7.
The Operating system (on the work PC where I encountered the above issue) is Windows 7 on x64.
Java platforms installed are:   1.8.0_211 _x86_64 and 1.8.0_211. (Version 8 Update 211 build 1.8.0_2111-b12).

I also tried all the google provided solutions (the _setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true_ command, I tried to add the certificates from the website manually to the java folder in the cacerts file because the website can be accessed from Chrome or Mozilla browsers, My company is not using any proxy servers but I also tried to specify no proxy in the Arduino IDE preferences window, I disabled with my IT the firewalls and the antivirus temporarily to see if it is blocked by it but the antivirus and the firewalls were not blocking it, and many more other "workarounds" found via google).
The solution for me was to return from versions 1.88 and 1.8.9 to 1.8.7.

Arduino IDE version 1.8.7 doesn't have any issues downloading the libraries and the boards from the  https://downloads.arduino.cc/packages/package_index.json

MattCosturos, 2019-05-03:
Hey MChima,

Arduino installs it's own jre in C:\Program Files (x86)\Arduino\java\ and that cacerts file needs to be updated!

So try to manually import the certs into the cacerts file in the JRE which was installed in the Arduino directory! (not in the global java directory!)

MChima, 2019-05-06:
Hi MattCosturos,

Thank you very much for the suggestion, I did not knew that Arduino uses his own java !
I tried to import them manually into Arduino IDE 1.8.9 and it did not worked, I get the same error when I try for example to open Board manager.

closed 2022-02-11 by MaSven