arduino/Arduino #7949: IDE 1.8.6 - SD-Lib "Datalogger" - segmentation fault

mikrocoder, 2018-08-29:
Hallo,

I have the following sporadic error. Memory is checked with Memtest and okay.
If I copy the sketch into a new window, no errors will occur.
With the older SD-Lib version of IDE 1.8.5 compiles without errors under the new IDE 1.8.6
Windows 10 64Bit

`Arduino: 1.8.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Worker\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Worker\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Worker\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10806 -build-path C:\Users\Worker\AppData\Local\Temp\arduino_build_196433 -warnings=all -build-cache C:\Users\Worker\AppData\Local\Temp\arduino_cache_166815 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Program Files (x86)\Arduino\libraries\SD\examples\Datalogger\Datalogger.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Worker\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Worker\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Worker\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10806 -build-path C:\Users\Worker\AppData\Local\Temp\arduino_build_196433 -warnings=all -build-cache C:\Users\Worker\AppData\Local\Temp\arduino_cache_166815 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Program Files (x86)\Arduino\libraries\SD\examples\Datalogger\Datalogger.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "-IC:\\Program Files (x86)\\Arduino\\libraries\\SD\\src" "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp" -o nul
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\SD\src\File.cpp
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\SD\src\SD.cpp
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\SD\src\utility\Sd2Card.cpp
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\SD\src\utility\SdFile.cpp
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\SD\src\utility\SdVolume.cpp
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "-IC:\\Program Files (x86)\\Arduino\\libraries\\SD\\src" "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp" -o "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Sketch wird kompiliert...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "-IC:\\Program Files (x86)\\Arduino\\libraries\\SD\\src" "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp" -o "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp.o"
Compiling libraries...
Compiling library "SPI"
Zuvor kompilierte Datei wird verwendet: C:\Users\Worker\AppData\Local\Temp\arduino_build_196433\libraries\SPI\SPI.cpp.o
Compiling library "SD"
Zuvor kompilierte Datei wird verwendet: C:\Users\Worker\AppData\Local\Temp\arduino_build_196433\libraries\SD\File.cpp.o
Zuvor kompilierte Datei wird verwendet: C:\Users\Worker\AppData\Local\Temp\arduino_build_196433\libraries\SD\SD.cpp.o
Zuvor kompilierte Datei wird verwendet: C:\Users\Worker\AppData\Local\Temp\arduino_build_196433\libraries\SD\utility\Sd2Card.cpp.o
Zuvor kompilierte Datei wird verwendet: C:\Users\Worker\AppData\Local\Temp\arduino_build_196433\libraries\SD\utility\SdFile.cpp.o
Zuvor kompilierte Datei wird verwendet: C:\Users\Worker\AppData\Local\Temp\arduino_build_196433\libraries\SD\utility\SdVolume.cpp.o
Compiling core...
Using precompiled core: C:\Users\Worker\AppData\Local\Temp\arduino_cache_166815\core\core_arduino_avr_mega_cpu_atmega2560_0c812875ac70eb4a9b385d8fb077f54c.a
Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433/Datalogger.ino.elf" "C:\\Users\\Worker\\AppData\\Local\\Temp\\arduino_build_196433\\sketch\\Datalogger.ino.cpp.o" "C:\\Users\\Wo [... cut ...]
compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Bibliothek SPI in Version 1.0 im Ordner: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI  wird verwendet
Bibliothek SD in Version 1.2.2 im Ordner: C:\Program Files (x86)\Arduino\libraries\SD  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.`

per1234, 2018-08-29:
A previous "Segmentation fault" issue report:
https://github.com/arduino/Arduino/issues/7946

I've also encountered sporadic errors similar to this while using the Arduino AVR Boards 1.6.20x beta releases. I was never able to produce a MCVE. The problem would go away after I restarted my computer. The problem did not occur when the same code was compiled with Arduino AVR Boards 1.6.21.

mikrocoder, 2018-09-05:
I have switched back to version v1.8.5 now

PaulStoffregen, 2018-09-05:
I tried it just now on Windows 10 64 bit with Arduino 1.8.6 using AVR core 1.6.22.

Clicked Verify a dozen times.  All worked perfectly.

[image]

mikrocoder, 2018-09-05:
I can't say otherwise how it is. v1.8.5 works without errors and 1.8.6 with sporadic errors with this example. I've never had a segmentation error before. Possibly a rare error in the 1.8.6 toolchain. I'm also not the only one with the segmentation fault problem. https://forum.arduino.cc/index.php?topic=566826.0

Additional question.  Why is the variable pathidx not initialized in SD.cpp?

qbahn21, 2018-09-19:
Confirm. I've also experienced such a sporadic errors with 1.8.6 toolchain. 1.8.5 works great!

per1234, 2018-11-29:
I'm going to start a list of the reports of this issue in case it helps with the investigation:
- https://arduino.stackexchange.com/q/58304
- https://forum.arduino.cc/index.php?topic=523894.msg3948770#msg3948770
- https://forum.arduino.cc/index.php?topic=541588.msg3691156#msg3691156
- https://forum.arduino.cc/index.php?topic=558353.msg3890444#msg3890444
- https://forum.arduino.cc/index.php?topic=565368.msg3852233#msg3852233
- https://forum.arduino.cc/index.php?topic=567551
- https://forum.arduino.cc/index.php?topic=567799.msg3869246#msg3869246
- https://forum.arduino.cc/index.php?topic=568281
- https://forum.arduino.cc/index.php?topic=568472
- https://forum.arduino.cc/index.php?topic=568731
- https://forum.arduino.cc/index.php?topic=571348
- https://forum.arduino.cc/index.php?topic=572746
- https://forum.arduino.cc/index.php?topic=573335
- https://forum.arduino.cc/index.php?topic=573924
- https://forum.arduino.cc/index.php?topic=574020
- https://forum.arduino.cc/index.php?topic=574672
- https://forum.arduino.cc/index.php?topic=575535
- https://forum.arduino.cc/index.php?topic=576328
- https://forum.arduino.cc/index.php?topic=577926
- https://forum.arduino.cc/index.php?topic=578034.msg3937187#msg3937187
- https://forum.arduino.cc/index.php?topic=578956.msg3942081#msg3942081
- https://forum.arduino.cc/index.php?topic=579689
- https://forum.arduino.cc/index.php?topic=579730
- https://forum.arduino.cc/index.php?topic=580092
- https://github.com/arduino/Arduino/issues/7946
- https://github.com/arduino/Arduino/issues/7949
- https://github.com/arduino/Arduino/issues/8037
- https://github.com/arduino/Arduino/issues/8045
- https://github.com/FastLED/FastLED/issues/646
- https://github.com/felis/USB_Host_Shield_2.0/issues/413
- https://github.com/Freenove/Freenove_Quadruped_Robot_Kit/issues/5
- https://github.com/JimBrown/MarlinTarantula/issues/56
- https://github.com/jp112sdl/HB-UNI-RGB-LED-CTRL/issues/2
- https://github.com/MarlinFirmware/Marlin/issues/11710
- https://github.com/MarlinFirmware/Marlin/issues/12048
- https://github.com/MarlinFirmware/Marlin/issues/12184#issuecomment-436090359
- https://github.com/MarlinFirmware/Marlin/issues/12320
- https://github.com/MarlinFirmware/Marlin/issues/12506
- https://github.com/MKFirmware/MK4duo/issues/585
- https://github.com/neu-rah/ArduinoMenu/issues/196#issuecomment-417939606
- https://github.com/olikraus/u8g2/issues/701

In all cases, the OS is Windows.

I'll continue to add to the list as additional reports come in.

facchinm, 2018-11-29:
@per1234 I'm investigating the issue by bisecting Atmel patches and vanilla gcc versions. It could take a bit. Reproducing consistently on all OSes with this sketch https://github.com/arduino/ArduinoCore-avr/issues/39 (when that will be fixed I think also this issue could be automatically solved).

facchinm, 2018-11-29:
Here's a json ready for testing for all host OS (http://downloads-02.arduino.cc/packages/package_avr_7.3.0_index.json). It will trigger an update to avr core **1.6.208.**

The build is based on vanilla gcc since Atmel patches for 5.4.0 break an internal sanity test when finalizing the build (didn't have the time to try to fix it yet). The missing Atmel patches are likely to break or limit support for ATTiny so **DON'T USE THIS THING IN PRODUCTION** :smile: 

However, all other boards should be supported (avr4809 too). This toolchain fixes arduino/ArduinoCore-avr#39 while I couldn't reproduce #7949 so I hope you can test it and report back.

@matthijskooijman do you still have the mass builder script with all the fancy statistics lying around? It would be cool to have a 4.9.2 vs 5.4.0 vs 7.3.0 :wink:

mikrocoder, 2018-11-30:
I have tested ArduinoCore version 1.6.23.
https://github.com/arduino/ArduinoCore-avr
and then compiled the Sketch SD-Datalogger countless times.
No segemantation errors so far.   👍

matthijskooijman, 2018-11-30:
@facchinm, Nice to see a vanilla gcc. If we can get that to work (possibly with some patches applied upstream), that would totally rock over having to wait for atmel all the time :-)

As for the builder script, it's here: https://github.com/matthijskooijman/arduino-mass-builder
I haven't used it anymore since I first wrote it, but I suspect it still works :-)

facchinm, 2018-12-03:
I've just updated the toolchain to `arduino4` with all (I hope :slightly_smiling_face: ) the Atmel patches included. AVR core gets updated to .209 ; compiling for 4809 is now possible, although the megaAVR core needs an extra patch (the toolchain version in use is hardoced in `platform.txt`).
@matthijskooijman I'll launch the mass builder soon, thanks for the link :wink:

per1234, 2018-12-03:
I've been hoping to check whether the new toolchain fixes the segmentation fault but I haven't been able to reproduce that bug even with Arduino AVR Boards 1.6.23, despite compiling multiple sketches that have caused it for me in the past on and off. It's really a pain of a bug.

I have been using Arduino AVR Boards 1.6.208 (and now 1.6.209) with no problems since it was published. I'll be sure to report if I discover anything of note.

mikrocoder, 2018-12-07:
I tested the datalogger example with IDE v1.8.8, couldn't provoke a segmentation error.  :-)

ImpulseAdventure, 2018-12-20:
Another validation datapoint: I had reported a similar crash with **lto**, **gcc 5.4.0** & **Windows** in https://github.com/arduino/ArduinoCore-avr/issues/55.

@facchinm -- upon retesting with the revised package json with AVR boards 1.6.208 & IDE 1.8.7, I'm happy to report that I no longer observe the compiler crash.

As a reference for others, here is the simple testcase I constructed for the bug, using a couple structs and an `extern "C"` wrapper: https://github.com/ImpulseAdventure/Arduino-LTO-Error

Psychokiller1888, 2019-01-08:
Same here, randomly happens. Since today impossible to compile at all anymore, where I could sporadically compile before

paynterf, 2019-02-10:
Same problem with V1.8.7 on Windows 10 home edition.  

```
Compiling debug version of 'StepperSpeedCtrl' for 'Arduino/Genuino Mega w/ ATmega2560 (Mega 2560)'
Build Folder: "file:///C:/Users/Frank/AppData/Local/Temp/VMBuilds/StepperSpeedCtrl/mega_atmega2560/Debug"
Additional Defines: VM_DEBUG;VM_DEBUG_ENABLE 1;VM_DEBUG_BANDWIDTH_THROTTLE_MS 50;VM_DEBUGGER_TYPE_HARDWARESERIAL 0;VM_DEBUGGER_TYPE_SOFTWARESERIAL 1;VM_DEBUGGER_TYPE_FASTSERIAL 2;VM_DEBUGGER_TYPE_USB 3;VM_DEBUGGER_TYPE_TEENSY 4;VM_DEBUGGER_TYPE_UART 5;VM_DEBUGGER_TYPE_USART 6;VM_DEBUGGER_TYPE_USBSERIAL 7;VM_DEBUGGER_TYPE_TTYUART 8;VM_DEBUGGER_TYPE_NET_CONSOLE 9;VM_DEBUGGER_TYPE_Uart 10;VM_DEBUGGER_TYPE_COSA 11;VM_DEBUGGER_TYPE_CDCSerialClass 12;VM_DEBUGGER_TYPE_HARDWARESERIAL1 13;VM_DEBUGGER_TYPE_HARDWARESERIAL2 14;VM_DEBUGGER_TYPE_HARDWARESERIAL3 15;VM_DEBUGGER_TYPE_NET_UDP 16;VM_DEBUGGER_TYPE_USBAPI 17;VM_DEBUGGER_TYPE_SERIALUSB 18;VM_DEBUGGER_TYPE_MS430_SERIAL_ 19;VM_DEBUGGER_TYPE VM_DEBUGGER_TYPE_HARDWARESERIAL;VM_DEBUG_BREAKPAUSE;
Architecture Tools: "file:///C:/Program%20Files%20(x86)/Arduino/hardware/tools/avr/bin/"
Api: 1221
Sketch Book: "file:///C:/Users/Frank/Documents/Arduino"
Sketch Include Paths
Core Include Paths
Include Path "file:///C:/Program%20Files%20(x86)/Arduino/hardware/arduino/avr/cores/arduino"
Include Path "file:///C:/Program%20Files%20(x86)/Arduino/hardware/arduino/avr/variants/mega"
 
Deep search for libraries ...
Using previous search results: C:\Users\Frank\AppData\Local\Temp\VMBuilds\StepperSpeedCtrl\mega_atmega2560\Debug\StepperSpeedCtrl.cpp
Using previous search results: C:\Users\Frank\AppData\Local\Temp\VMBuilds\StepperSpeedCtrl\mega_atmega2560\Debug\StepperSpeedCtrl.cpp
Using previous search results: C:\Users\Frank\AppData\Local\Temp\VMBuilds\StepperSpeedCtrl\mega_atmega2560\Debug\StepperSpeedCtrl.cpp
Using previous search results: C:\Program Files (x86)\Arduino\libraries\Stepper\src\Stepper.cpp
"C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DVM_DEBUG_BREAKPAUSE -DVM_DEBUGGER_TYPE=VM_DEBUGGER_TYPE_HARDWARESERIAL -DVM_DEBUGGER_TYPE_MS430_SERIAL_=19 -DVM_DEBUGGER_TYPE_SERIALUSB=18 -DVM_DEBUGGER_TYPE_USBAPI=17 -DVM_DEBUGGER_TYPE_NET_UDP=16 -DVM_DEBUGGER_TYPE_HARDWARESERIAL3=15 -DVM_DEBUGGER_TYPE_HARDWARESERIAL2=14 -DVM_DEBUGGER_TYPE_HARDWARESERIAL1=13 -DVM_DEBUGGER_TYPE_CDCSerialClass=12 -DVM_DEBUGGER_TYPE_COSA=11 -DVM_DEBUGGER_TYPE_Uart=10 -DVM_DEBUGGER_TYPE_NET_CONSOLE=9 -DVM_DEBUGGER_TYPE_TTYUART=8 -DVM_DEBUGGER_TYPE_USBSERIAL=7 -DVM_DEBUGGER_TYPE_USART=6 -DVM_DEBUGGER_TYPE_UART=5 -DVM_DEBUGGER_TYPE_TEENSY=4 -DVM_DEBUGGER_TYPE_USB=3 -DVM_DEBUGGER_TYPE_FASTSERIAL=2 -DVM_DEBUGGER_TYPE_SOFTWARESERIAL=1 -DVM_DEBUGGER_TYPE_HARDWARESERIAL=0 -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=50 -DVM_DEBUG_ENABLE=1 -DVM_DEBUG   -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" -I"C:\Program Files (x86)\Arduino\libraries\Stepper\src" -I"c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers\VM_DBG" -I"c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers\VM_DBG\utility" "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers\VM_DBG\VM_DBG.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DVM_DEBUG_BREAKPAUSE -DVM_DEBUGGER_TYPE=VM_DEBUGGER_TYPE_HARDWARESERIAL -DVM_DEBUGGER_TYPE_MS430_SERIAL_=19 -DVM_DEBUGGER_TYPE_SERIALUSB=18 -DVM_DEBUGGER_TYPE_USBAPI=17 -DVM_DEBUGGER_TYPE_NET_UDP=16 -DVM_DEBUGGER_TYPE_HARDWARESERIAL3=15 -DVM_DEBUGGER_TYPE_HARDWARESERIAL2=14 -DVM_DEBUGGER_TYPE_HARDWARESERIAL1=13 -DVM_DEBUGGER_TYPE_CDCSerialClass=12 -DVM_DEBUGGER_TYPE_COSA=11 -DVM_DEBUGGER_TYPE_Uart=10 -DVM_DEBUGGER_TYPE_NET_CONSOLE=9 -DVM_DEBUGGER_TYPE_TTYUART=8 -DVM_DEBUGGER_TYPE_USBSERIAL=7 -DVM_DEBUGGER_TYPE_USART=6 -DVM_DEBUGGER_TYPE_UART=5 -DVM_DEBUGGER_TYPE_TEENSY=4 -DVM_DEBUGGER_TYPE_USB=3 -DVM_DEBUGGER_TYPE_FASTSERIAL=2 -DVM_DEBUGGER_TYPE_SOFTWARESERIAL=1 -DVM_DEBUGGER_TYPE_HARDWARESERIAL=0 -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=50 -DVM_DEBUG_ENABLE=1 -DVM_DEBUG   -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" -I"C:\Program Files (x86)\Arduino\libraries\Stepper\src" -I"c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers\VM_DBG" -I"c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers\VM_DBG\utility" "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers\VM_DBG\VM_mem_check.c" -o "nul"
[... cut ...]
	//delay(500);  //2 readings per sec to start
	//delay(100); 
	delay(50);  

	//now set the motor speed and move some steps.  Too many steps and it won't be responsive
	//to speed changes.  Too few and it will be too choppy.
	myStepper.setSpeed(rpm);
	myStepper.step(10);
	numloops++;
}

Might be code-length dependent.  I couldn't get it to fail with the Blink sketch, failed about 1 out of 5 with this sketch, and 8/10 on my 3K+ line sketch

Just started happening yesterday and today.  Not sure why.

```

ImpulseAdventure, 2019-05-15:
FYI @facchinm & @per1234 -- in case it is helpful to have a very minimal test sketch to reproduce the LTO failure in IDE 1.8.9 w/o any other dependencies, please see: [Arduino-LTO-Error](https://github.com/ImpulseAdventure/Arduino-LTO-Error).

For the benefit of others looking for a solution (without rolling back to an older version):
- The workaround of installing `package_avr_7.3.0_index.json` with latest **Arduino AVR Boards** (1.6.209) resolves the issue:
  - Add [http://downloads.arduino.cc/packages/package_avr_7.3.0_index.json](http://downloads.arduino.cc/packages/package_avr_7.3.0_index.json) to the Preferences -> Settings -> Additional Boards Manager URLs
  - In Tools -> Board -> Boards Manager, locate **Arduino AVR Boards** and click on Update. This should be reflected as version `1.6.209`.

MartyMacGyver, 2019-09-08:
It's been just over a year now. Is anyone officially working on this issue so it can be resolved in the main Arduino toolchain versus the "don't use in production" workarounds?

mikrocoder, 2019-09-08:
Well, guys, I wasn't taken seriously back then. Neither here nor in the Arduino Forum. Can only say uses a current toolchain.

MartyMacGyver, 2019-09-21:
@mikrocoder Looks like that's changed now - give 1.8.10 a try, and be sure to update your board compilers via the boards manager.

ImpulseAdventure, 2019-09-21:
Fantastic! **Arduino IDE 1.8.10** (with the default **Arduino AVR Boards 1.8.1**) appears to have resolved the LTO issues I had previously observed in my testcases.

@facchinm & the rest of the Arduino team -- thank you very much for the fix!

mikrocoder, 2019-09-21:
Nice. I'm about to test it. What was fixed is unfortunately not in the release note. Although I now use a current toolchain.

MartyMacGyver, 2019-09-21:
I think you need to download the latest board package via the boards manager after the install, but it's from the official source now, isn't it? Or does it depend on the alternate source still?

facchinm, 2019-09-23:
@MartyMacGyver it's official now; the relevant part of the changelog is [here](https://github.com/arduino/Arduino/blob/master/build/shared/revisions.txt#L24) (should backport to avr repo)

closed 2019-09-23 by facchinm

MartyMacGyver, 2019-09-23:
Great news! Thank you.