arduino/Arduino #7955: Menus defined in boards.txt have no defined ordering

sleemanj, 2018-08-31:
If one defines menus in a boards.txt there is no defined or even reliable or repeatable ordering for how those menus are sorted in the Tools menu

Example, if one defines menus thusly

    menu.version    = Processor Version
    menu.clock      = Processor Speed
    menu.boot       = Use Bootloader
    menu.core       = Millis, Tone Support
    menu.millis_ppm  = Millis Accuracy
    menu.print      = Print Support
    menu.serial     = Serial Support
    menu.lto        = Link Time Optimisation (1.6.11+ only)
    menu.bodlevel   = Brown-out Detection Level
    menu.advclk     = Override Clock Source
    menu.advfrq     = Override Frequency
    menu.advups     = Override Upload Speed

then one might reasonably expect them to be shown in this order, however the actual results appear largely at-random with no discernible pattern I can see, not alpha, not first appearance in the file... I think it's just pretty much random.

[image]

facchinm, 2018-08-31:
Hi @sleemanj,
this is due to HashSet not being intrinsically ordered in Java.
I've created a very simple PR to test this https://github.com/arduino/Arduino/pull/7956 ; let me know if it fixes the issue so we can merge it safely, thanks!

sleemanj, 2018-08-31:
Compiled and tested Linux x64 Java 8.  

The ordering is now alpha by menu title and seems to work without issue.  

That's certainly an improvement over random - core authors can easily fudge menu titles with a prefixed index number "01. Blah Menu", "02. Foo Menu", or they can also be a bit kludgy and use a sequences of zero width space characters before each menu whichseems to work (1 zero-width space = 1st item, 2 zero-width spaces = 2nd item...) also, just a bit hard to see what you're doing when coding it :-)

Ideally of course instead of alpha it would just preserve the order the menus were defined in boards.txt, but that might be more complicated.

cmaglie, 2018-09-04:
@sleemanj could you try this build https://github.com/arduino/Arduino/pull/7956#issuecomment-418148851? I've used a `LinkedHashSet` instead of a `TreeSet` and it seems to work as intended. Java magics :-)

PR #7956 (2018-09-04): Avoid custom menus random order
Fixes #7955
commit 3dfe8591 (2018-09-04): Preserve boards.txt custom menu order.

Fix #7955
closed 2018-09-04 by cmaglie