arduino/Arduino #7970: Arduino IDE will not open on Linux Ubuntu 18.04

CodingCaio, 2018-09-06:
Suddently Arduino IDE 1.8.6 will not start. I've been working on it without problems for days until now. I added WiFiManager library into my libraries using the Library Manager, then I tried to compile it but it threw an error saying "Can't find <WifiManager.h>" even thought I just had added it. Then I closed the IDE to restart it but I haven't been able to open it again. The spash screen shows and it closes right after showing "Preparing boards...".

Here is the error that I get:

```
	:~/arduino-1.8.6$ ./arduino
	Picked up JAVA_TOOL_OPTIONS: 
	Error parsing library.index:Unexpected end-of-input: expected close marker for Object (start marker at [Source: (FileInputStream); line: 45890, column: 5])
	 at [Source: (FileInputStream); line: 45901, column: 5664]
	 at [Source: (FileInputStream); line: 45901, column: 7] (through reference chain: com.fasterxml.jackson.module.mrbean.generated.cc.arduino.contributions.libraries.LibrariesIndex["libraries"]->java.util.ArrayList[2164])
	java.lang.NullPointerException
		at cc.arduino.contributions.libraries.LibrariesIndexer.rescanLibraries(LibrariesIndexer.java:140)
		at cc.arduino.contributions.libraries.LibrariesIndexer.setLibrariesFolders(LibrariesIndexer.java:109)
		at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:677)
		at processing.app.Base.onBoardOrPortChange(Base.java:1313)
		at processing.app.Base$12.actionPerformed(Base.java:1540)
		at processing.app.Base.filterVisibilityOfSubsequentBoardMenus(Base.java:1579)
		at processing.app.Base.access$000(Base.java:83)
		at processing.app.Base$11.actionPerformed(Base.java:1512)
		at processing.app.Base.rebuildBoardsMenu(Base.java:1490)
		at processing.app.Base.<init>(Base.java:267)
		at processing.app.Base.main(Base.java:145)
```

I tried removing and installing the Arduino IDE multiple times and rebooting the system. Any more ideas?

PaulStoffregen, 2018-09-06:
Duplicate of #7937.  Fixed by #7964.

To recover, either download and run the 2nd arduinobot build on #7964.  Or run this in a terminal

    rm ~.arduino15/library_index.json

You may need to open the library manager twice to see the correct library list.

closed 2018-09-06 by PaulStoffregen

CodingCaio, 2018-09-06:
Sorry for the duplicate. I did search, but the error is so generic that I got many unrelated results. @PaulStoffregen , that worked great! Thanks man! I wasn't aware of the hidden .arduino15, I had looked for a json file. Thanks a lot!

per1234, 2018-09-06:
This same variant ("Unexpected end-of-input") of https://github.com/arduino/Arduino/issues/7937 was reported on the Arduino forum (http://forum.arduino.cc/index.php?topic=565839) and I got that user to send me the problematic library_index.json file. I couldn't find anything wrong with the file they provided and I had no problems starting Arduino IDE 1.8.6 with that file in place. However, the workaround of removing the problematic library_index.json did solve it for them so I believe that, although this error is a little different than https://github.com/arduino/Arduino/issues/7937, https://github.com/arduino/Arduino/pull/7964 will still solve it.

CodingCaio, 2018-09-06:
I don't know if this helps but my initial error was:

`Gtk-Message: Failed to load module "canberra-gtk-module"`

It took me to [this thread](https://askubuntu.com/questions/342202/failed-to-load-module-canberra-gtk-module-but-already-installed) and some people got the issue resolved by installing the modules in the answer. I installed the modules and then the error became the one I showed in my original post above.

cmaglie, 2018-09-06:
> Really looking forward to a 1.8.7 release....

we'll release Tuesday Sep,11

MasterCATZ, 2018-11-03:
root@aio:/SnapRaidRW/DATA/Downloads/arduino-1.8.7# ./arduino
Picked up JAVA_TOOL_OPTIONS: 
mmap failed for CEN and END part of zip file
Error occurred during initialization of VM
java/lang/ClassNotFoundException: error in opening JAR file /SnapRaidRW/DATA/Downloads/arduino-1.8.7/java/lib/rt.jar
root@aio:/SnapRaidRW/DATA/Downloads/arduino-1.8.7# 

root@aio:/SnapRaidRW/DATA/Downloads/arduino-1.8.7# lsb_release -a
LSB Version:	security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic
root@aio:/SnapRaidRW/DATA/Downloads/arduino-1.8.7# uname -a
Linux aio 4.18.16-041816-generic #201810200431 SMP Sat Oct 20 08:33:06 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

MasterCATZ, 2019-01-01:
/arduino-1.8.8$ ./arduino
Picked up JAVA_TOOL_OPTIONS: 
mmap failed for CEN and END part of zip file
Error occurred during initialization of VM
java/lang/ClassNotFoundException: error in opening JAR file

facchinm, 2019-01-02:
@MasterCATZ it looks like you need to redownload the IDE zip since it's corrupt (make sure the checksum matches [this](https://downloads.arduino.cc/arduino-1.8.8.md5sum.txt)