arduino/Arduino #7978: Java JDK & JRE versions mismatched on Macintosh

PaulStoffregen, 2018-09-09:
Arduino 1.8.6 bundles Java JRE 8u171 on Mac, Linux and Windows.

However, the Mac version appears to be compiled with Java JDK 8u121, much older than the JRE used.  The Linux and Windows builds appear to be built using JDK 8u171 which matches the bundled JRE.

I just ran some quick tests compiling pde.jar.  JDK 8u121 & 8u171 produce slightly different bytecode output for classes Base, Editor, EditorTab, SketchController, PdeKeywords, SketchTextArea, Archiver, FixEncoding, ZipDeflater and UpdateCheck.

So far this doesn't seem to cause anything bad to happen, but the Mac build system probably should be updated to JDK 8u171.

facchinm, 2018-09-10:
Hi Paul, thanks for reporting! I just updated the JDK to the right version on the OSX build machine so we should be ok; feel free to reopen the issue if we are still missing something (we'll provide you a WIP version of 1.8.7 to make sure everything is fine).

closed 2018-09-10 by facchinm

PaulStoffregen, 2018-09-12:
@facchinm - Downloaded the 1.8.7 release files just now.  Looks like the Mac version was again built with the old JDK 8u121.

PaulStoffregen, 2018-09-12:
The nightly file on the same page looks like it was built with JDK 8u171.  But the official release looks like JDK 8u121.

facchinm, 2018-09-12:
Thanks for spotting this, I found the misconfiguration on Jenkins side; the rebuild is ongoing and should replace the wrong one soonish

PaulStoffregen, 2018-09-12:
Confirmed fixed.  Downloaded the new file just now, shasum a0da9e7d3da5bc8ee66c8b7ad0adddb862376a99.  It appears to be build with JDK 8u171.  :)