arduino/Arduino #7979: IDE crashes on start when sketchbook folder path contains spaces or is symbolic link

happenpappen, 2018-09-09:
This happens to me in IDE versions 1.8.5 to 1.9.0 BETA on Linux:

When configuring a sketchbook folder with spaces in the path like "/home/user/my documents/sketchbook" it will crash with this error on the next start:

Picked up JAVA_TOOL_OPTIONS: 
java.lang.AbstractMethodError
        at processing.app.Editor.addTools(Editor.java:868)
        at processing.app.Editor.buildToolsMenu(Editor.java:723)
        at processing.app.Editor.buildMenuBar(Editor.java:536)
        at processing.app.Editor.<init>(Editor.java:286)
        at processing.app.Base.handleOpen(Base.java:842)
        at processing.app.Base.handleOpen(Base.java:828)
        at processing.app.Base.handleOpen(Base.java:824)
        at processing.app.Base.handleNew(Base.java:763)
        at processing.app.Base.<init>(Base.java:469)
        at processing.app.Base.main(Base.java:145)

The same will happen if you change the sketchbook folder to a symbolic link pointing to a folder.

I'm quite sure this did not happen in older versions since i've used a symbolic link as sketchbook folder for a long time.

happenpappen, 2018-09-10:
Hi @facchinm ,

thank you for your reply. I removed the outdated plugins from sketchbook/tools and the IDE is starting now. Maybe it is possible to extend the IDE to be more "verbose" about an error like this?

kind regards,

Christoph

closed 2018-09-10 by happenpappen

facchinm, 2018-09-11:
The IDE has a lot of places where exceptions are thrown without the relevant information to debug the issue; unfortunately we don't have enough manpower to do it... Anyway it would be a prefect Google Summer of Code project :wink: