arduino/Arduino #7980: wdt_disable does not work on attiny85

zoranbosnjak, 2018-09-09:
It looks like WDT is still enabled, when the processor comes out of WDT reset. The problem is that `wdt_disable()` (called from the `setup` function) does not disable the WDT. 

Observation from the example code below:
After power up, 5 regular LED blinks (until the first WDT reset), then it looks like the processor enters infinite reset loop (very fast LED blinking).

Expected behaviour:
5 LED blinks, WDT reset, start from the beginning... 5 LED blinks...

There is a workaround to the problem if a custom `wdt_off` function is used instead of original `wdt_disable`. In this case, the program works as expected.

Tested with arduino-1.8.5 and attiny85.

```C
#include <avr/wdt.h>

#define LED 4   // PB4, pin3

volatile int cnt=0;

// this function replaces wdt_disable()
void wdt_off() {
  wdt_reset();
  MCUSR = 0x00;
  WDTCR |= (1<<WDCE) | (1<<WDE);
  WDTCR = 0x00;
}

void setup() {
  wdt_disable();
  // wdt_off();
  pinMode(LED, OUTPUT);
}

void loop() {
  digitalWrite(LED, HIGH);
  delay(100);
  digitalWrite(LED, LOW);
  delay(900);

  cnt++;
  if (cnt >= 5) {
    // setup watchdog to reset CPU
    wdt_enable(WDTO_15MS);
    while(1) {}
  }
}
```

facchinm, 2018-09-10:
Hi @zoranbosnjak ,
please open an issue on the the proper core repository (probably this one https://github.com/SpenceKonde/ATTinyCore ) since the problem is not specific to the Arduino IDE. Thanks!

closed 2018-09-10 by facchinm

zoranbosnjak, 2018-09-10:
OK, thanks for the info. I have reported this problem here:
https://github.com/SpenceKonde/ATTinyCore/issues/242