arduino/Arduino #7982: Error downloading https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

BILLGLENN, 2018-09-10:
This keeps rearing its ugly head but I've never seen someone follow up with the solution. Please give a helping hand after reading this summary. It contains specific facts pertaining and hopefully sufficient for a pro to be able to solve. I have seen lots of guessing -- none of it useful. What, if one pleases, is the specific cause of this shared problem across an assortment of platforms? :::

The Windows reinstall will work for a while. I have been battling this issue for 3 weeks now. At one point, I installed my cloned C:/ drive that I had created 2 weeks before this problem began. After swapping out the problematic C:/ with the clone, everything worked just great. So, I thought whatever was the problem was now banished from my system. Not so fast! About 1 week later, all of a sudden, the EXACT problem was back:

Error downloading https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
Error downloading https://www.adafruit.com/package_adafruit_index.json

Keep in mind that NOTHING about the internet network or connectivity has been modified. Everyday, just routine usage of the Arduino UI and the same Adafruit board project was conducted. Then, one day, one just wakes up, turns on the computer, loads Arduino and BAM! --- the Adafruit boards will not load and thus there is no access to them. It's a complete disruption of what has been a long and prosperous development cycle. But no more. I am literally at my wits end trying to guess why this boards manager cannot get the Adafruit boards. They are the only ones that are blocked for some unknown reason. And, just for those who may be thinking viruses/malware etc., a full scan has been accomplished and there is nothing there. I hope there is some IT techy who will figure this one out. (It occurs across platforms) and it's only the Adafruit, third party files that are inaccessible. Thanks in advance.

PaulStoffregen, 2018-09-10:
At first glance this might look like a duplicate of #7937, but I believe what we're seeing here is a separate issue.

Perhaps the library manager is overwriting the original good JSON file when it begins downloading an update.  If the new download fails or ends up with bad data, is the original good data lost?

Maybe the download should be done to a temporary file.  Then it could be verified, ideally against a SHA2 hash, but at the very least by checking the JSON data parses correctly.  The previously good index should only be replaced after the download verifies as good data.

pasindu-sandima, 2019-09-03:
I had the same issue and I fixed it doing the following steps.

1. Open Arduino IDE
2. Go to preferences
3. Click on the link to edit preferences.txt ( This will take you to `C:\Users\$Username\AppData\Local\Arduino15`)
4. Delete the temp file and the json file from that directory
5. Delete the package in the packages folder
6. Delete staging files related to that in the staging folder
7. Restart the device and add the link again and try

[image]
[image]
[image]

italex33, 2020-03-01:
My problem was that by accident I set the whole directory to read only.