arduino/Arduino #7985: Printing without delay will cause confusion.

UBTEDU, 2018-09-11:
The second time you open the serial monitor, the data will be confused.

[image]

matthijskooijman, 2018-09-11:
I'm assuming you mean the `?L` being printed, and the continuous 1's are what you expect? You're being a bit vague about what the problem is exactly.

I suspect this is because opening the serial monitor resets the Arduino. If that happens when it is halfway through sending a byte, that byte will get mangled, which could be what you're seeing. This does not explain more than one mangled byte, though.

The behaviour I'm describing is pretty much impossible to fix. Since the serial drivers will buffer data, I believe you cannot ensure you only get data from after the reset (there is no way to tell when a byte in the buffer was generated on the Arduino) AFAICS.

mikaelpatel, 2018-09-11:
@UBTEDU What is the expected behavior here? 

I guess that is would be possible to view the serial monitor as just a way to visualize the serial channel and that the serial channel would allows be connected to the board (after upload). This would give some additional "logic" to reset the board (as @matthijskooijman has explained). 

Cheers!

PaulStoffregen, 2018-09-11:
Indeed this is a problem with all hardware serial communication.  If the stream is non-stop back-to-back bytes transmitted, the receiver can not reliably find the first start bit.  If it begins looking for incoming data during the middle of a byte, any of the high-to-low edges can be mistaken for the start bit.  Then it will end within the middle of the next byte and being looking for a start bit again.  There are some data patters where the receiver never manages to sync to the intended start & stop bits.  Of course, if the transmitter ever leaves the line idle for 9+ bit times, the receiver will always manage to find the next start bit.  This is simply how asynchronous serial communication works.

matthijskooijman, 2018-09-11:
Ah, that explains why more than one byte is corrupted. Thanks for pointing that out :-)

As I think there is nothing that can be done to fix this, I'm closing this issue. If anyone has any further ideas, feel free to leave additional comments.

closed 2018-09-11 by matthijskooijman