arduino/Arduino #7995: Content Spoofing or Text Injection

tarun776, 2018-09-15:
**Hello i want to report a text injection and a miss configuration of the login page which can be used for maliciuos activty.**

Description :

An attacker is able to inject his own text into error page and can fool the victim to visit his own malicious site and other things

**Vulnerable link POC -**
https://auth.arduino.cc/login/?error=!!!ATENTION!&error_description=YOUR%20ACCOUNT%20HAS%20BEEN%20HACKED.%20PLEASE%20CONTACT%20on%201234567890.%20and%20EMAIL%20:%20ATTACKER@GMALI.COM%20|%20visit%20:%20www.attacker.com
[image]

I can redirect users for malicious activity.

I hope you will fix it ASAP

And Please provide reward and hof

Thanks for reading

matteosuppo, 2018-09-19:
We are about to deploy a fix

tarun776, 2018-09-19:
ok thank u

Any Bounty ??

tarun776, 2018-09-20:
reply please ??

tarun776, 2018-09-20:
any bounty ??

per1234, 2018-09-21:
This appears to have been resolved so I'm going to close it:
[image]
I can't speak for Arduino regarding a bounty but I think that contributing to a free open source project is a reward in itself.

closed 2018-09-21 by per1234