arduino/Arduino #7997: Empty sketch not compiling on Arch Linux using glibc (Uno board)

MrYakobo, 2018-09-15:
Greetings

It seems like the latest update in either glibc or Arduino broke something. I can't compile the default sketch anymore. I'm probably missing something... Below is the verbose output of the compiler.

```
Arduino: 1.8.5 (Linux), Board: "Arduino/Genuino Uno"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jakob/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/jakob/.arduino15/packages -libraries /home/jakob/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path /tmp/arduino_build_278585 -warnings=none -build-cache /tmp/arduino_cache_280061 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/jakob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9 -prefs=runtime.tools.arduinoOTA.path=/home/jakob/.arduino15/packages/arduino/tools/arduinoOTA/1.1.1 -prefs=runtime.tools.avr-gcc.path=/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2 -verbose /tmp/untitled891068964.tmp/sketch_sep15a/sketch_sep15a.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jakob/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/jakob/.arduino15/packages -libraries /home/jakob/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path /tmp/arduino_build_278585 -warnings=none -build-cache /tmp/arduino_cache_280061 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/jakob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9 -prefs=runtime.tools.arduinoOTA.path=/home/jakob/.arduino15/packages/arduino/tools/arduinoOTA/1.1.1 -prefs=runtime.tools.avr-gcc.path=/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2 -verbose /tmp/untitled891068964.tmp/sketch_sep15a/sketch_sep15a.ino
Using board 'uno' from platform in folder: /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21
Using core 'arduino' from platform in folder: /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21
Detecting libraries used...
"/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino" "-I/home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/variants/standard" "/tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino" "-I/home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/variants/standard" "/tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp" -o "/tmp/arduino_build_278585/preproc/ctags_target_for_gcc_minus_e.cpp"
"/usr/bin/arduino-ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_278585/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino" "-I/home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/variants/standard" "/tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp" -o "/tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp.o"
In file included from /usr/include/stdlib.h:55:0,
                 from /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino/Arduino.h:23,
                 from /tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp:1:
/usr/include/bits/floatn.h:74:70: error: unknown machine mode '__TC__'
 typedef _Complex float __cfloat128 __attribute__ ((__mode__ (__TC__)));
                                                                      ^
/usr/include/bits/floatn.h:86:9: error: '__float128' does not name a type
 typedef __float128 _Float128;
         ^
In file included from /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino/Arduino.h:26:0,
                 from /tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp:1:
/usr/include/math.h:415:21: error: '_Float128' was not declared in this scope
 # define _Mdouble_  _Float128
                     ^
/usr/include/math.h:283:48: note: in definition of macro '__MATHDECL_1'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
                                                ^
/usr/include/bits/mathcalls-helper-functions.h:21:36: note: in expansion of macro '_Mdouble_'
 __MATHDECL_1 (int, __fpclassify,, (_Mdouble_ __value))
                                    ^
In file included from /usr/include/features.h:428:0,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino/Arduino.h:23,
                 from /tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp:1:
/usr/include/sys/cdefs.h:61:20: error: expected ',' or ';' before 'throw'
 #   define __THROW throw ()
                    ^
/usr/include/math.h:283:53: note: in expansion of macro '__THROW'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
                                                     ^
/usr/include/bits/mathcalls-helper-functions.h:21:1: note: in expansion of macro '__MATHDECL_1'
 __MATHDECL_1 (int, __fpclassify,, (_Mdouble_ __value))
 ^
In file included from /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino/Arduino.h:26:0,
                 from /tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp:1:
/usr/include/math.h:415:21: error: '_Float128' was not declared in this scope
 # define _Mdouble_  _Float128
                     ^
/usr/include/math.h:283:48: note: in definition of macro '__MATHDECL_1'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
                                                ^
/usr/include/bits/mathcalls-helper-functions.h:25:33: note: in expansion of macro '_Mdouble_'
 __MATHDECL_1 (int, __signbit,, (_Mdouble_ __value))
                                 ^
In file included from /usr/include/features.h:428:0,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino/Arduino.h:23,
                 from /tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp:1:
/usr/include/sys/cdefs.h:61:20: error: expected ',' or ';' before 'throw'
 #   define __THROW throw ()
[... 171 lines cut ...]
 extern int strcmp_PF(const char *s1, uint_farptr_t s2) __ATTR_PURE__;
                                      ^
/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:1682:39: error: 'uint_farptr_t' has not been declared
 extern int strncmp_PF(const char *s1, uint_farptr_t s2, size_t n) __ATTR_PURE__;
                                       ^
/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:1698:42: error: 'uint_farptr_t' has not been declared
 extern int strcasecmp_PF(const char *s1, uint_farptr_t s2) __ATTR_PURE__;
                                          ^
/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:1716:43: error: 'uint_farptr_t' has not been declared
 extern int strncasecmp_PF(const char *s1, uint_farptr_t s2, size_t n) __ATTR_PURE__;
                                           ^
[... cut ...]
/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:1744:37: error: 'uint_farptr_t' has not been declared
 extern size_t strlcpy_PF(char *dst, uint_farptr_t src, size_t siz);
                                     ^
/home/jakob/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:1760:36: error: 'uint_farptr_t' has not been declared
 extern int memcmp_PF(const void *, uint_farptr_t, size_t) __ATTR_PURE__;
                                    ^
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

Cheers

MrYakobo, 2018-09-16:
I should add that if the Arduino IDE shipped with some static glibc library this would not be a problem. Cheers

per1234, 2018-09-16:
The first problem I see is that you're using Arduino IDE 1.8.5. The latest release of the Arduino IDE is 1.8.7. 1.8.5 is a year out of date. You should always test using the latest [hourly build](https://www.arduino.cc/en/Main/Software#hourly) before reporting an issue to be sure that the problem has not always been solved.

This sort of output shows the problem:
```
In file included from /usr/include/stdlib.h:55:0,
                 from /home/jakob/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino/Arduino.h:23,
                 from /tmp/arduino_build_278585/sketch/sketch_sep15a.ino.cpp:1:
/usr/include/bits/floatn.h:74:70: error: unknown machine mode '__TC__'
 typedef _Complex float __cfloat128 __attribute__ ((__mode__ (__TC__)));
                                                                      ^
/usr/include/bits/floatn.h:86:9: error: '__float128' does not name a type
 typedef __float128 _Float128;
         ^
```
A properly working official version of the Arduino IDE should never be including files from /usr/include.

I suspect both problems are caused by you using an unofficial, modified version of the Arduino IDE installed from the package manager.

MrYakobo, 2018-09-16:
Ok, downloaded the hourly build, log below:

```

MrYakobo, 2018-09-17:
avr-g++

```
Using built-in specs.
Reading specs from /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=./avr-g++
COLLECT_LTO_WRAPPER=/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../libexec/gcc/avr/5.4.0/lto-wrapper
Target: avr
Configured with: ../gcc/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir --disable-nls --disable-libssp --disable-libada --disable-shared --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 5.4.0 (GCC) 
```

avr-gcc

```
Using built-in specs.
Reading specs from /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=./avr-gcc
COLLECT_LTO_WRAPPER=/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../libexec/gcc/avr/5.4.0/lto-wrapper
Target: avr
Configured with: ../gcc/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir --disable-nls --disable-libssp --disable-libada --disable-shared --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 5.4.0 (GCC) 
```

avr-cpp

```
Using built-in specs.
Reading specs from /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=./avr-cpp
Target: avr
Configured with: ../gcc/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir --disable-nls --disable-libssp --disable-libada --disable-shared --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 5.4.0 (GCC) 
COLLECT_GCC_OPTIONS='-E' '-v' '-specs=device-specs/specs-avr2'
 /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../libexec/gcc/avr/5.4.0/cc1 -E -quiet -v -iprefix /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/ - -mn-flash=1 -mskip-bug
ignoring nonexistent directory "/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/sys-include"
ignoring duplicate directory "/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/../../lib/gcc/avr/5.4.0/include"
ignoring duplicate directory "/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/../../lib/gcc/avr/5.4.0/include-fixed"
ignoring nonexistent directory "/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/../../lib/gcc/avr/5.4.0/../../../../avr/sys-include"
ignoring duplicate directory "/home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/../../lib/gcc/avr/5.4.0/../../../../avr/include"
ignoring duplicate directory "/usr/include/"
  as it is a non-system directory that duplicates a system directory
#include "..." search starts here:
#include <...> search starts here:
 /usr/include/
 /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/include
 /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/include-fixed
 /home/jakob/Downloads/arduino-nightly-linux64/arduino-nightly/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/include
End of search list.
```

cc1

```
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/lib/gcc/avr/5.4.0/include"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/lib/gcc/avr/5.4.0/include-fixed"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/sys-include"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/include"
ignoring duplicate directory "/usr/include/"
  as it is a non-system directory that duplicates a system directory
#include "..." search starts here:
#include <...> search starts here:
 /usr/include/
End of search list.
```

cc1plus

```
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/include/c++/5.4.0"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/include/c++/5.4.0/avr"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/include/c++/5.4.0/backward"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/lib/gcc/avr/5.4.0/include"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/lib/gcc/avr/5.4.0/include-fixed"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/sys-include"
ignoring nonexistent directory "/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir/avr/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/include/
End of search list.
```

matthijskooijman, 2018-09-17:
Ok, so gcc somehow puts `/usr/include` in the include path. I was under the impression that these paths are bulitin, but apparently these can also be [taken from environment variables](https://serverfault.com/a/147599). Could you check if you have `C_INCLUDE_PATH` or `CPLUS_INCLUDE_PATH` set? If not, can you paste the output of the `env` command to show your full environment in case there's something else in there?

@MrYakobo

MrYakobo, 2018-09-18:
Good find. `C_INCLUDE_PATH` is set to `/usr/include`. `CPLUS_INCLUDE_PATH` is not defined.

Output of `env` is below. Worth noting is `CPATH: /usr/include`

```
AWT_TOOLKIT=MToolkit
BROWSER=firefox
COLORTERM=truecolor
CPATH=/usr/include/
C_INCLUDE_PATH=/usr/include/
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
DISPLAY=:0
EDITOR=vim
FREETYPE_PROPERTIES=truetype:interpreter-version=40
GOPATH=/home/jakob/go
HOME=/home/jakob
LANG=en_US.UTF-8
LOGNAME=jakob
MAIL=/var/spool/mail/jakob
MANPAGER=less -s -M +Gg
MOZ_USE_XINPUT2=1
OSTYPE=linux-gnu
PAGER=less
PATH=/home/jakob/.autojump/bin:/home/jakob/bin:/home/jakob/.npm/bin:/usr/bin:/home/jakob/.gem/ruby/2.5.0/bin/:/home/jakob/go/bin:/home/jakob/.cargo/bin/:/home/jakob/.autojump/bin:/usr/local/sbin:/usr/local/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
PWD=/home/jakob
QT_DIR=/home/jakob/Qt/5.11.1/
QT_QPA_PLATFORMTHEME=qt5ct
QT_VERSION=5.11.1
SHELL=/usr/bin/fish
SHLVL=4
TERM=xterm-termite
USER=jakob
VTE_VERSION=5002
WINDOWID=25165827
WINDOWPATH=1
XAUTHORITY=/home/jakob/.Xauthority
XDG_CURRENT_DESKTOP=GNOME
XDG_RUNTIME_DIR=/run/user/1000
XDG_SEAT=seat0
XDG_SESSION_ID=1
XDG_VTNR=1
_JAVA_AWT_WM_NONREPARENTING=1
```

closed 2018-09-18 by MrYakobo

matthijskooijman, 2018-09-18:
Thanks for reporting back, good to hear it is solved. I wonder if arduino-builder should handle this case (e.g. clear those variables), though it probably is useful to keep them in some cases as well, so I guess it is best to just keep things as they are.