arduino/Arduino #7999: arduino offcial cannot find IDE shortkey

mygoodboy, 2018-09-16:
example

Ctrl+T | Auto   Format
Ctrl+G | Find   Next

please tell me how to know  IDE document about shortcut in https://www.arduino.cc
other people share good content but it isn't latest version

per1234, 2018-09-16:
I don't think it's necessary to write documentation for these shortcuts because they are already documented right there in the UI menus:
[image]
[image]

mygoodboy, 2018-09-16:
ctrl+D not exist  in IDE bar but it can use and exist
I want to know all shortcuts 

thank you!!

per1234, 2018-09-16:
That's a more compelling argument. Another undocumented shortcut:
- **Ctrl+{** - Jump to the matching bracket

There's also **Ctrl+MouseWheel** to change font size and **Ctrl+Tab** to advance tabs, but those are pretty standard.

I suppose the appropriate place to document the shortcuts would be:
https://www.arduino.cc/en/Guide/Environment