arduino/Arduino #8004: liblistSerials 1.4.1 Linux ARM problem

per1234, 2018-09-16:
Using Arduino IDE 1.8.7 Hourly Build 2018/09/12 02:23 / Arduino IDE 1.8.7 on Raspberry Pi 3 B+, Raspbian Stretch

Starting the IDE fails with this error:
```
pi@raspberrypi:~/ArduinoIDE/arduino-nightly $ bash arduino
Picked up JAVA_TOOL_OPTIONS: 
java.lang.UnsatisfiedLinkError: /home/pi/ArduinoIDE/arduino-nightly/lib/liblistSerialsj.so: /home/pi/ArduinoIDE/arduino-nightly/lib/liblistSerialsj.so: cannot open shared object file: No such file or directory
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
	at java.lang.Runtime.load0(Runtime.java:809)
	at java.lang.System.load(System.java:1086)
	at processing.app.Platform.loadLib(Platform.java:162)
	at processing.app.Platform.<clinit>(Platform.java:157)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at processing.app.BaseNoGui.initPlatform(BaseNoGui.java:515)
	at processing.app.Base.<init>(Base.java:200)
	at processing.app.Base.main(Base.java:145)
/home/pi/ArduinoIDE/arduino-nightly/lib/liblistSerialsj.so: /home/pi/ArduinoIDE/arduino-nightly/lib/liblistSerialsj.so: cannot open shared object file: No such file or directory
Cannot load native library /home/pi/ArduinoIDE/arduino-nightly/lib/liblistSerialsj.so
The program has terminated!
```
The problem goes away after replacing ~/ArduinoIDE/arduino-nightly/lib/liblistSerialsj.so with the one from Arduino IDE 1.8.6 (liblistSerials 1.4.0).

Originally reported at:
http://forum.arduino.cc/index.php?topic=568797

victordfb, 2020-02-07:
Version 1.8.11, on raspberry and I still have the problem. Is it possible to reopen this issue?

matthijskooijman, 2020-02-07:
Yup, we can. Are you seeing exactly the same issue? Maybe you can post the error you get? Also, do you have `liblistSerialsj.so` in your Arduino install? Where did you install from? I just downloaded the 32-bit ARM build from arduino.cc, and that one does have the file.

rtorrero, 2020-02-11:
I was having the issue today and I found out that the community package I had used for openSUSE Tumbleweed had a broken symlink to my system's `liblistSerialsj.so`.

Manually fixed the link and it's working again.

facchinm, 2020-02-11:
Closing as fixed upstream. @rtorrero can you notify openSUSE developers that their package is broken? Thanks

closed 2020-02-11 by facchinm

rtorrero, 2020-02-17:
> Closing as fixed upstream. @rtorrero can you notify openSUSE developers that their package is broken? Thanks

These are community packages, so not really officially maintained by openSUSE. That said, just changing the provider did the trick too.