arduino/Arduino #8006: Board Manager: problems with OpenCR - select version or remove

KurtE, 2018-09-17:
I am running Arduino 1.8.7 with Teensyduino 1.44-Beta2 (probably not related). 
Windows 10 64 bit, installed Arduino using zip file. 

I have several boards installed by the board manager including the Robotis OpenCR
```
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://tician.github.io/arbotix/package_trossen_arbotix_index.json
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCM9.04/master/arduino/opencm_release/package_opencm9.04_index.json
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json
```
A few days ago, after installing 1.8.7, when I started up Arduino it said that some of my boards had updated.  So I updated OpenCR to 1.3.2... 

Yesterday I tried building again using 1.3.2 and it failed.  There was an issue in their new release so I tried going back to 1.3.1, there were a few strange messages during this in the message area.  And when I tried to build using it.  It complained about OpenCR was not a board or some such thing, it also looked like it was partially still trying to use 1.3.2 files, but could not find boards.txt...

Last night got a response back from Robotis saying they fixed it and I should remove the board and reinstall.  So this morning I tried uninstalling and received the following errors:

```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:213)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:717)
	at processing.app.Base.onBoardOrPortChange(Base.java:1313)
	at processing.app.Base$12.actionPerformed(Base.java:1540)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1490)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1378)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:211)
	... 1 more
java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:717)
	at processing.app.Base.onBoardOrPortChange(Base.java:1313)
	at processing.app.Base$12.actionPerformed(Base.java:1540)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1490)
	at processing.app.Base.openBoardsManager(Base.java:1396)
	at processing.app.Base$10.actionPerformed(Base.java:1418)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2238)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2296)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4897)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4534)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4475)
	at java.awt.Container.dispatchEventImpl(Container.java:2282)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
Sorry if this is a known issue, I did not see any obvious duplicates, but I could have easily missed them.

Kurt

facchinm, 2018-09-17:
Hi @KurtE ,
this is probably related with the not-so-clear history of the json; in fact, the IDE fails often if an older release is being re-released or, worse, deleted. It's a known issue we are trying to address with `arduino-cli` integration.

In the meantime, anytime something bad happens with jsons it's better to remove the extracted folder from Arduino15 folder and redownload everything from scratch.

KurtE, 2018-09-17:
Note: I then tried to update again to 1.3.2, brought up the board manager, it then had an error about the CRC did not match....  

Restarted Arduino, now it appears to fault... Will try removing folder, and see if that helps.

Thanks

KurtE, 2018-09-17:
OpenCR package was updated again, and I now was able to download the updated 1.3.2

Not sure if I should go ahead and close or if someone will want to diagnose further on why a corrupted board description kept 1.8.6 and 1.8.7, but could open 1.8.5