arduino/Arduino #8009: High CPU load - Bug of code editor?

PascalKoe, 2018-09-18:
When I have Arduino running in the background (doesn't matter if device is connected or not) the IDE has a ~33% load on my CPU.
If I loose the focus of the code editor within the IDE by for example opening the serial monitor, clicking the little down arrow to create a new tab or using the upper menu, the load on the CPU drops to ~0%. After closing the serial monitor (or closing the menus, ...) and typing in the code editor the CPU load stays low. Everything fine like that.
But now if I loose focus of the IDE Window and regain focus on the window, the CPU load rises back up again to ~33% load.

Arduino IDE Version 1.8.7 
Windows 10 Pro 64bit Version 10.0.17134 Build 17134 
Intel Core i5-4570
8GB DDR3 RAM

Only thing special about my system is that I'm using WinToUsb (Bootloader installed on an external M.2 drive, Windows is on a VHD on the external drive which gets mounted and started from).

Does this bug exist on other systems?

Thank you

PascalKoe, 2018-09-19:
Hi @facchinm 
Tester Versions 1.8.7 to 1.8.4 and 1.9.0-beta-81.
Version 1.8.6 and higher (including the 1.9.0 beta) have the same issue.

Other modifications I made to the IDE are...
... changing the font to Consolas but changing back to Monospaced doesn't make a difference
... installed the ESP32 Core using the boardmanager
... installed VisualMicro

Additionally I do not have Java installed on my machine.
Maybe I will try to install Java and see if that makes a difference.

PascalKoe, 2018-09-19:
Just build the IDE myself like described on the wiki page. Works without the bug :)
The hourly build (17 September 2018 10:43:2 GMT) still has the bug.. Shouldn't the hourly and the self build be identically?
I was able to recreate the bug on a different installation by booting a Windows 10 Edge developer vhdx, available [here](https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/) (MSEdge on Windows 10, HyperV).

facchinm, 2018-09-20:
@PascalKoe I've been able to reproduce the problem and it looks related with JRE u171 . Moving to u181 seems to fix the high CPU load while in background issue. I'm updating the bundled jre in hourly, it would be nice if you could try it in some hours to make sure the bug has gone.

pierre83, 2018-09-22:
Hello,
Please note that I have the same problem with my Windows Vista but not on my Windows 10 PC.
The JAVA process is using 50% of my old CPU.
Trying your new bundle
Pierre

facchinm, 2018-09-24:
Hi @pierre83 ,
the hourly version available here (https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip) contains JREu181 ; would you mind testing it? Thanks!

PascalKoe, 2018-09-24:
Sorry for the delay.
Yes, the bug seems fixed in the current nightly :)
Thank you!
Pascal

pierre83, 2018-09-24:
Hi !
Thank you all, yesterday, I downloaded the nightly (with JREu181) and the problem is gone !

Pierre

sergbrest, 2018-09-25:
I also confirm the problem is gone ! Ver. nightly 2018/09/20 02:27

pierre83, 2018-09-25:
Hello,
The CPU problem is gone but I encountered a new problem, which is not related to the JAVA version.
It's impossible to load thru "ArduinoAsISP" on a barebone ATMEGA328P, because the "
arduinoasisp.protocol=stk500v1" and "arduinoasisp.program.protocol=stk500v1" lines in the programmers.txt file are not to "stk500v1" like here  but with "arduino"

facchinm, 2018-09-25:
@pierre83 this is another problem, please open another issue to properly keep track of it, thanks.

facchinm, 2018-09-25:
Closing as fixed by updating bundled JREs on Jenkins

closed 2018-09-25 by facchinm

DonpK, 2018-09-25:
I have also noted the high CPU usage and can reproduce it as described in this thread: https://github.com/arduino/Arduino/issues/7505 , that is, open the IDE and then open the Serial Monitor and disconnect the USB connection. the CPU usage jumps to ca. 50% for the process: C:\Program Files (x86)\Arduino\java\bin\javaw.exe 

I'm a relatively new Arduino user and certainly have no knowledge of the underlying IDE java code, but I'm happy to provide any more information if it's useful. As a newbie should I try the nightly mentioned above or am I better off waiting until a fix is incorporated into the next official version?

per1234, 2018-09-25:
>I'm happy to provide any more information if it's useful.

It's only useful if you test the nightly build and find the problem still occurs. Problems you encounter with the stable releases may have already been fixed since that time.

>As a newbie should I try the nightly mentioned above or am I better off waiting until a fix is incorporated into the next official version?

The nightly builds are intended to be used only for beta testing. If you want to do beta testing then you should definitely be using the latest nightly build. Otherwise, you should only use the nightly build if it fixes a serious problem that occurs for you with the stable releases.

DonpK, 2018-09-26:
Thanks, now that I know how to work around it, I'll wait for the next general release where, hopefully, it will be addressed.

d1deco, 2018-11-20:
Nightly build  1.8.8  2018/11/16 07:33  did not solve the problem of high cpu, after removing USB.
I added printscreen of taskmanager  with some comment inside.
Will also add procmon output, but need  some info about setting up  filter.
[image]

facchinm, 2018-11-21:
@d1deco this is a different problem and is being tackled by another PR https://github.com/arduino/Arduino/pull/8046 . Could you test it and report if it solved your problem?

d1deco, 2018-12-07:
I just  tried  arduino-PR-beta1.9-BUILD-90. Removing USB (serial COM6 in my case)  no longer give High CPU. 
 Java version  Java(TM) SE Runtime Environment (build 1.8.0_191-b12)
New problem arise.
If serial monitor is open, upload is impossible.  
last lines of  compilation ... 
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM6
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
error: Failed to open COM6
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_open failed

facchinm, 2018-12-10:
@d1deco the beta includes patch https://github.com/arduino/Arduino/pull/8046 so it is expected to solve the high cpu issue on serial disconnection.
However, there should be no problem in uploading; can you post the verbose output of your compilation and upload (for example, which esp8266 board are you using? which core version?). Thanks!

d1deco, 2018-12-11:
I apologise. Problem I saw  is due to a wacky usb3 port on this elder E5530 notebook. No problem when changing into a usb2 type port.