arduino/Arduino #8010: macOS High Sierra - Ports not showing for Arduino Mega 2560

wjlow, 2018-09-18:
This is a genuine Arduino. When I plug in the USB cable I can see the Arduino light up. However, I'm unable to select any ports in Arduino IDE, the only one there is the Bluetooth Incoming Port.

I tried doing `ls /dev` before and after the Arduino is connected and I get the same output.

I also don't see the device in System Report -> USB. However, I did manage to get it show up once. Must've been from one of the many drivers I tried installing :/

Can anyone please help?

Would it work if I installed Ubuntu onto this machine instead?

per1234, 2018-09-19:
This topic is more appropriate for the [Arduino Forum](http://forum.arduino.cc/). If you haven't already, please start a post on the forum. I'm sure we'll be able to help you with your problem over there. If we do end up determining the problem is caused by a bug in the Arduino IDE, I'll reopen this issue.

closed 2018-09-19 by per1234

PaulStoffregen, 2018-09-19:
FWIW, I can confirm Arduino Mega does indeed work with MacOS High Sierra (10.13.6) using the driver Apple includes with the system.