arduino/Arduino #8013: Please inlcude micro-Maqueen-Arduino-Library

kd8bxp, 2018-09-19:
https://github.com/kd8bxp/micro-Maqueen-Arduino-Library
A simple library for use with the DF Robot micro:Maqueen Robot and BBC Micro:Bit
thank you

per1234, 2018-09-19:
For inclusion in the Arduino Library Manager index, it's required to comply with the library.properties `name` field specifications:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.

The `:` character in your `name` value: 
```
name=micro:Maqueen
```
is not allowed. Please change your library.properties file to use a specification compliant name and then either create a new tag/release or else redo the existing tag.

Once that's done, comment here so we know we can proceed with adding your library to the index.

kd8bxp, 2018-09-19:
done with a new tag 1.0.1

cmaglie, 2018-09-20:
done!

closed 2018-09-20 by cmaglie