arduino/Arduino #8015: Please include https://github.com/mcci-catena/arduino-lmic in the library list

terrillmoore, 2018-09-19:
Please add MCCI's LoRaWAN library to the list: https://github.com/mcci-catena/arduino-lmic.

cmaglie, 2018-09-20:
There is already a library called `IBM LMIC framework`:
https://github.com/matthijskooijman/arduino-lmic.git

You should change the name of the library in this way:
1) update the `name` field in `library.properties`
2) commit & push
3) delete and redo the git release tag.

/cc @matthijskooijman

matthijskooijman, 2018-09-20:
TBH, I've not been investing enough time in my version of the library lately, so there are a few versions floating around. I'm still planning to do an overhaul and cleanup of the library so we can hopefully integrate the versions into a single version again, but it's just not a priority for me. Not sure what the best approach for now is. It seems my version is still the defacto standard version, even though it might not be the most complete version...

terrillmoore, 2018-09-20:
@cmaglie Thanks for your feedback.

Sorry to be confused!

Are you suggesting that I rename the  library to be identically named to https://github.com/matthijskooijman/arduino-lmic.git ? Or should I rename it to be more different? Or something else?

BTW... this library is tailored specifically for MCCI devices and has a lot of extra features that are required by MCCI users and other MCCI libraries. The extra features make it not interchangeable with @matthijskooijman at this point, although there is strong family resemblance. When I refactored to add Japan. AS923, and India support, we really lost the ability to share patches, because things moved to different files.

cmaglie, 2018-09-20:
The requirement is that the name should be something *different* from `IBM LMIC framework`, choose whatever you feel right for this library,

> The extra features make it not interchangeable with @matthijskooijman at this point, although there is strong family resemblance

Integrating your changes into Matthijs' lib would have been the best option (for the devs and the users, having a single code base to maintain and bugfix is always better), that's why I've /cc @matthijskooijman here, but if it's not feasible you can just change the library name in library.properties.

terrillmoore, 2018-09-20:
This is why I'm confused. Please check library.properties again on the tag.  I already have: 

https://github.com/mcci-catena/arduino-lmic/blob/2e76a81c72e0d57c9c1c7b9654b92a46175a269b/library.properties#L1

```
name=MCCI LoRaWAN LMIC library
```

Where do you see the other name?