arduino/Arduino #8021: Feature Request: Don't close app when every sketch is closed

Jones-S, 2018-09-21:
Dear Arduino Devs,
This is a thing that bothered me since years. I finally make the effort to ask you to improve your software: I really dislike the behaviour that the application closes whenever I close all my sketches. I sometimes close them accidentally and after taht I have to wait quite a while to restart the app.
Would it be possible to adapt the behaviour like we all know it from other apps. 
chrome .e.g would not quit, just because I close all my windows...

This would be a great improvement and I know a lot of people who actually think the same.
Thank you for sharing your thoughts about this.

cheers

J

facchinm, 2018-09-21:
Hi @Jones-S ,
would you mind testing https://github.com/arduino/Arduino/pull/8023 when the build is ready? It's not yet bug free but it could be a beginning.

Jones-S, 2018-09-23:
Oh that was quick! I downloaded the mac build and opened the app. So far it does what I want it to. It stays open. 👏🏼 for that.
Unfortunately I'm currently abroad and I don't have any arduinos with me. anyway I found a first bug: When no sketch is open (but the app is) and I want to open up a recent item: `File` -> `Open recent`
It does not open the file. The command only works if there is already a sketch open.
Should I report this as a new issue?
Cheers

facchinm, 2018-09-24:
@Jones-S I'm closing the issue as duplicate of #936 (which contains more information).
In the meantime I should have fixed the remaining bugs you found so I'd love if you could test the new build as soon as it is available  on #8023 . Thanks!

closed 2018-09-24 by facchinm