arduino/Arduino #8022: Please add new library: AstroMech

weckbach, 2018-09-21:
Hi,

I would like to add my new library to be available for Arduino:

https://github.com/weckbach/AstroMech

AstroMech is a serial-over-audio protocol to broadcast data over audio.

Thanks

per1234, 2018-09-21:
From the Library Manager FAQ:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-is-the-library-list-generated
>A library is not valid when:
...
>- it contains symlinks

Your library contains a symlink:
https://github.com/weckbach/AstroMech/blob/0.1/src/src
so that will need to be removed before your library can be added to the Library Manager index.

I think there might be another problem too. Apparently the `url` field in library.properties is not allowed to be left blank:
https://github.com/arduino/Arduino/issues/8000#issuecomment-421800498
And yours is blank:
https://github.com/weckbach/AstroMech/blob/0.1/library.properties#L8
Anyway, it seems quite simple and helpful to put the URL of your repository in that field.

Please resolve these issues and then either create a new tag/release or redo the existing tag. Once that's done, please comment here so we know we can proceed.

weckbach, 2018-09-21:
Thanks for the quick response. The issues are fixed.