arduino/Arduino #8032: Arduino nightly 2018/09/20 02:27 : "ArduinoAsISP" upload fail

pierre83, 2018-09-26:
Hi,
Uploading a barebone ATMEga328P with a NANO (my case) is impossible because in the file "programmers.txt" the following lines seems to be wrong:
"arduinoasisp.protocol=arduino" & "arduinoasisp.program.protocol=arduino" and should be
"arduinoasisp.protocol=stk500v1 & "arduinoasisp.program.protocol=stk500v1"
When uploading, the symptom is:
         Using Port                    : COM4
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15

and further:
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

per1234, 2018-09-26:
Have you checked whether the problem is resolved by changing the programmer definition back to using the `stk500v1` protocol?

Related commit: https://github.com/arduino/ArduinoCore-avr/commit/b084848f2eaf9ccb3ac9a64ac5492d91df4706bf

pierre83, 2018-09-26:
Yes, the problem is resolved by changing these 2 lines from Arduino to stk500v1

per1234, 2018-10-06:
Here's the PR where the protocol was changed, which has a lot of information on the reason for doing so: https://github.com/arduino/Arduino/pull/3687
I had some trouble finding it because the PR is still open, but the commit from the PR was pulled into Arduino AVR Boards.

There is another report of this causing a problem here:
http://forum.arduino.cc/index.php?topic=569766.msg3895180#msg3895180

per1234, 2018-12-12:
This issue has come up multiple times on the forum since the release of Arduino AVR Boards 1.6.22. I wonder if it wouldn't be better to revert https://github.com/arduino/ArduinoCore-avr/commit/b084848f2eaf9ccb3ac9a64ac5492d91df4706bf and add a new "Arduino as ISP (ATmega32U4)" programmer that uses the `arduino` protocol? Anyone have thoughts on that?

I made a [3rd party hardware package](https://github.com/per1234/ATmega32U4asISP) to add such a programmer years ago and have kept an eye out for people having trouble using ATmega32U4-based boards for Arduino as ISP ever since. That issue has rarely come up. So it seems to me that https://github.com/arduino/ArduinoCore-avr/commit/b084848f2eaf9ccb3ac9a64ac5492d91df4706bf is causing more trouble than it solves. I do like that it expands support for Arduino as ISP.

facchinm, 2018-12-12:
@per1234 totally agree, I'm taking care of it.

facchinm, 2018-12-19:
@per1234 I totally approve your solution, reverting to `stk500v1` protocol for all boards except 32u4 and creating another "programmer" with `arduino` protocol. Would you mind filing a PR to https://github.com/arduino/ArduinoCore-avr/ ? Thanks you so much for your invaluable contribution!

closed 2018-12-19 by facchinm