arduino/Arduino #8033: code enhancement

Heymann03, 2018-09-26:
I'm trying to use the following code for sketch. Need some suggestions on how to make the code running for the following sketch.

//sketch of the circuit board//
[image]

//following is the code file for the above sketch//

[Transmitter_and_Receiver_source_code.zip](https://github.com/arduino/Arduino/files/2420495/Transmitter_and_Receiver_source_code.zip)

per1234, 2018-09-26:
This issue tracker is only to be used to report bugs or feature requests. This is **NOT** an appropriate place to post this sort of thing. The appropriate place is the Arduino Forum. I'm sure we'll be able to help you over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2018-09-26 by per1234

Heymann03, 2018-09-28:
> If you're using Teensy, probably best to ask on this forum.
>
> https://forum.pjrc.com/forums/2-Project-Guidance

Thanks for the information.