arduino/Arduino #8034: --install-boards esp8266:esp8266:2.4.2 reports Selected board is not available

veleek, 2018-09-26:
I'm running the following command `& "C:\Program Files (x86)\Arduino\arduino_debug.exe" --install-boards esp8266:esp8266:2.4.2` from a powershell prompt and the installation fails with the error "Selected board is not available".

I have verified using Fiddler that it downloads the package definition from the expected location (`http://arduino.esp8266.com/stable/package_esp8266com_index.json`) and the package appears to be valid and it definitely contains a version 2.4.2, but the install just fails.

If I specify version 2.3.0 it also fails with the same error.

If I don't specify a version at all (just `esp8266:esp8266`) then it succeeds and it installs a package version 2.3.0 but I don't know where it's getting it from.

veleek, 2018-09-26:
Note: the `/stable` URL is simple a redirect to: http://arduino.esp8266.com/versions/2.4.2/package_esp8266com_index.json.

veleek, 2018-09-27:
Additional info, after digging around I determined that I had both the `/stable/package...` URL **AND** the `/versions/2.3.0/package_...` URL specified in the *Additional Board Manager URLs* configuration setting.  The 2.3.0 URL was specified first, so when I excluded version from the parameter that was the one that it chose to install.

I removed the 2.3.0 path and now if I exclude version it installs 2.4.2 but it still fails if I explicitly provide the version with the same error.

facchinm, 2018-10-05:
The regression has been introduced with https://github.com/arduino/Arduino/commit/3092e03becc932073566cb82f4d9bb7cdf3ce13d , https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Base.java#L308 call to `VersionHelper.valueOf(boardToInstallParts[2]).toString()` now returns `Optional[$versionNumber]` . 
@cmaglie fixing `toString()` should be enough right?

veleek, 2018-10-05:
FYI: I also see similar errors when attempting to install libraries using the Library Manager (from the VSCode plugin).  Is there a chance that this same code was modified in both the board manager and library manager?

cmaglie, 2018-10-08:
@veleek 
yes, that's the same problem, I'm going to add a fix on top of #8069

commit 6a1c1f4d (2018-10-15): Invalid use of toString in Optional<Version>

It should be used on the actual Version.

Fix #8034
closed 2018-10-15 by cmaglie