arduino/Arduino #8042: Error on a https://www.arduino.cc/en/Tutorial/Knob

benpaddlejones, 2018-09-30:
In the code Pin should either be Analogue pins A0, A1, A2 etc for bother servo and potentiometer (and diagram updated) or the maths need to be changed to pwm values. 

Below is changes to using Analogue pins

/*
 Controlling a servo position using a potentiometer (variable resistor)
 by Michal Rinott <http://people.interaction-ivrea.it/m.rinott>

 modified on 8 Nov 2013
 by Scott Fitzgerald
 http://www.arduino.cc/en/Tutorial/Knob
*/

#include <Servo.h>

Servo myservo;  // create servo object to control a servo

int potpin = A0;  // analog pin used to connect the potentiometer
int val;    // variable to read the value from the analog pin

void setup() {
  myservo.attach(A2);  // attaches the servo on pin 9 to the servo object
}

void loop() {
  val = analogRead(potpin);            // reads the value of the potentiometer (value between 0 and 1023)
  val = map(val, 0, 1023, 0, 180);     // scale it to use it with the servo (value between 0 and 180)
  myservo.write(val);                  // sets the servo position according to the scaled value
  delay(15);                           // waits for the servo to get there
}

per1234, 2018-09-30:
>`int potpin = A0;  // analog pin used to connect the potentiometer`

All the official Arduino cores allow you to pass either the analog channel number or pin number to `analogRead()`. Thus, `analogRead(0)` is functionally the same as `analogRead(A0)`. So this definitely can not be considered an error.

However, the `A0` pin name may be more clear because that's what's marked on the Arduino boards. The use of the channel number 0 might lead someone to think they should use the pin marked 0. I notice that the [`analogRead()` reference page](https://www.arduino.cc/reference/en/language/functions/analog-io/analogread/) also doesn't use the A*n* pin names.

>`myservo.attach(A2);  // attaches the servo on pin 9 to the servo object`

Why do you think it's necessary to change this from pin 9?

benpaddlejones, 2018-09-30:
Yes I assumed pin 0 was digital pin 0 not analog pin A0.

Correct me if I'm wrong but the PWM pins only do a analogWrite and the scale is between 0-255. So the read of the Potentiometer from PIN 9 won't work and if if did work the math for `val` would be wrong
` val = map(val, 0, 1023, 0, 180);` is wrong should be ` val = map(val, 0, 255, 0, 180);`.

Code provided works perfectly when A0 and A2 pins are used as both do analogueRead and analogueWrite on a scale of 0-1023 (tested with official Arduino UNO).

Ben

closed 2021-03-29 by per1234