arduino/Arduino #8048: Warn for assignment in if() upon compilation

robzach, 2018-10-01:
I often see students write `if (x = y)` when they mean `if (x == y)`. An understandable error; even a prof of C++ taking one of my intro classes made this same mistake in his code and wasn't able to see it.

Apparently `gcc` can be made to warn of assignments in `if` tests, by using the `-Wall` flag at compilation. The warning can be avoided by using an extra pair of parentheses to demonstrate programmer's intention, i.e. `if ((x = y))`.

As many users of the Arduino IDE are newer programmers and unlikely to intend assignments this way, I recommend that the warning for assignments inside of `if()` tests be made visible in the console.

PaulStoffregen, 2018-10-01:
Did you try this in File > Preferences?

[image]

robzach, 2018-10-02:
Thanks, Paul! 

I had overlooked that preference, and when I switched it to *All*, on compilation it threw exactly the right warning, helpfully painted orange to make it jump out: `warning: suggest parentheses around assignment used as truth value [-Wparentheses]`.

I'll now suggest setting that preference as one of the initial IDE setup steps for my students.

closed 2018-10-02 by robzach