arduino/Arduino #8051: Library: https://github.com/Blake-Tourneur/sevenSeg

Blake-Tourneur, 2018-10-02:
New library for library manager request, called sevenSeg.
Link: https://github.com/Blake-Tourneur/sevenSeg

per1234, 2018-10-02:
There is already a library in the Library Manager index named SevenSeg and each library must have a unique name, [regardless of case](https://github.com/arduino/Arduino/issues/7887#issuecomment-415033718). Please change the `name` value in your library.properties to a unique value and then either create a new release/tag or redo the existing one.

Once that's done, please comment here so we know we can proceed.

Blake-Tourneur, 2018-10-02:
I fixed it. New library link: https://github.com/Blake-Tourneur/singleSevSeg

per1234, 2018-10-02:
At the time of the v1.0.0 release, it still had the library.properties `name` value sevSeg. You need to either redo the release or create a new one with the unique name.

Please comment when that's resolved.

Blake-Tourneur, 2018-10-02:
Done

Blake-Tourneur, 2018-10-02:
The new release is v1.0.0, I deleted the old one

cmaglie, 2018-10-09:
done!

closed 2018-10-09 by cmaglie