arduino/Arduino #8052: macOS: Cannot upload using USBtinyISP

fpgaminer, 2018-10-03:
Tested on nightly: 28 September 2018 3:2:58 GMT
OS: macOS 10.14

**The Bug**
Attempting to upload a sketch using USBtinyISP as the selected programmer will always fail on macOS with the error "Please select a Port before Upload".  This is a bug because, as far as I understand, USBtinyISP is a USB device, not serial, so it is impossible to select a Port for it (this is at least true of the Adafruit Trinket).

This is a regression bug; uploading works fine on 1.6.13.

According to another party this bug is not present on Windows.

**Steps to reproduce:**

- Write an LED blink sketch (or any sketch)
- Select Tools->Board->Adafruit Trinket (ATtiny85 @ 8MHz)
- Select Tools->Programmer->USBtinyISP
- Click Upload

**Error Message:**

```
Please select a Port before Upload
```

**Expected Behavior:**

Arduino IDE should not require a Port to be selected when the programmer is USBtinyISP.

**Opinion:**

Looking through the code, it seems this was a "feature" added some point after v1.6.  Before, the code didn't care if there was a serial port or not and would just blunder forward through `newUploader` anyway (with `boardPort` left as `null`).  The correct fix, I guess, is for `getUploaderByPreferences` to be aware of which Programmer is selected and whether it needs a serial port or not.  Alternatively this feature could simply be rolled back (I've found no way to work around it; so it breaks Arduino 1.8 on macOS).

facchinm, 2018-10-04:
It looks like the bug starts from https://github.com/arduino/Arduino/commit/4ccf9bbd906607195e1aa1c4b0853dd6bc2c7f69 (included in #7958)
@PaulStoffregen the best way to tackle this could be parsing the upload rule in advance to decide if it's ok to fail for the missing serial port, otherwise I think we should revert to the old behaviour.
@fpgaminer the patch is contained in both 1.8.6 and 1.8.7 so the newest stable version you can use at the moment is 1.8.5 .

PaulStoffregen, 2018-10-04:
Opps, sorry for this.  Looking at it now.

I don't personally own a Trinket or USBtinyISP programmer.  I can get them if needed, but that will take several days.

Before we fully revert this, please consider the Arduino forum would repeatedly get the same new user question, where avrdude would try to access COM1.  We really should find a way to fix this, but avoid the extremely common case where first time users get stuck only because the default was COM1 and they use Windows and their PC really does have a COM1 port (usually on a motherboard header without any cable connected to the outside world).

PaulStoffregen, 2018-10-04:
@fpgaminer - I'm trying to follow your steps to reproduce the problem.  But I'm not able to do step 2:

Select Tools->Board->Adafruit Trinket (ATtiny85 @ 8MHz)

It does not appear in Tools > Boards.  I can not find any installation for it in Tools > Boards > Boards Manager.

Perhaps at some point you added another URL to the "Additional Boards Manager URLs" in File > Preferences?  Any chance you could look at that list and give me the URL needed to be able to install support for Trinket?

per1234, 2018-10-04:
@PaulStoffregen here it is:
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
It's in the "Adafruit AVR Boards" package.

You don't need the hardware to reproduce the bug.

PaulStoffregen, 2018-10-04:
Confirmed, I can reproduce the problem here on a Linux system.

Still looking for a clean way to fix this and prevent the extremely common case of avrdude cryptic (and very slow) messages when COM1 exists but isn't connected to anything.

fpgaminer, 2018-10-04:
Thank you for taking a look at this bug!

I'm not familiar enough with the codebase to contribute meaningfully beyond the bug report, but I'm curious; is `getUploaderByPreferences` able to get information about the currently selected programmer?  If so, I noticed that in the packaged `programmers.txt` a `communication` option is defined for most programmers specifying either `serial` or `usb`.  Perhaps that value could be checked, and if it's not `serial` then don't require that a serial port is selected?

ladyada, 2018-10-15:
@PaulStoffregen hiya will  you be able to take a look at this bug? some customers are asking me about it. perhaps for now i can tell them to just use arduino 1.8.6

ladyada, 2018-10-15:
FYI this also would affect Arduino Gemma because it uses the same technique

PaulStoffregen, 2018-10-23:
Sorry about the delay... I haven't forgotten about this issue, but at the moment I'm bringing up new hardware that's consuming all my dev cycles.  Hope to work on this in a week or so.  Please feel free to ping me if the silence goes on too long, or if there are any signs Arduino might be preparing for 1.8.8 (I'm as much in the dark as anyone regarding when they intend to publish non-beta releases).

I took the liberty of adding this to the 1.8.8 milestone list.

In the meantime, the 2 workarounds are to select *any* port from the Ports menu, or on a PC without any ports to select, use 1.8.6 or 1.8.5.

ajquick, 2018-11-01:
Just came across this problem myself. Had to roll back to 1.8.6 in order to get the USBtinyISP to work.

jjok, 2018-11-07:
_Me too_ with a Trinket on Linux.

cmaglie, 2018-11-09:
I'm looking at this one.

cmaglie, 2018-11-14:
Fixed by #8194

closed 2018-11-14 by cmaglie

FredCailloux, 2019-04-11:
I had the same problem with my Arduino installation until I tried this solution which worked perfectly for me. See: per1234 commented on Oct 4, 2018 , up here in this blog. 
Be prepared to redo any important configuration of your Arduino installation for this solution involve to replace the whole file "preferences.txt" . Once deleted Arduino detected my TinyProgrammer and I could upload my sketch no problem. Thanks per1234

TheCrazyT, 2019-11-10:
Is it normal to have this bug on version 1.8.11 (hourly build 2019/10/29/ 10:33) on windows?
The problem is that i have no serial interfaces() and use "AVRISP mkII".
Avrdude itself works with " -Pusb" but the arduino IDE fails to upload.
Deleting preferences.txt didn't help,too.