arduino/Arduino #8061: [Library Manager] Library not updating?

xoseperez, 2018-10-05:
Hi, the fauxmoESP library is in the manager but it has not been updated to the latest version (3.0.0) even though it was updated 20 days ago. Is there any reason for this?

This is the repo: https://bitbucket.org/xoseperez/fauxmoesp

facchinm, 2018-10-05:
Hi @xoseperez ,
the problem reported is
```
JOB 1627 - 2018/10/05 17:29:19 path=bitbucket.org/xoseperez, archivename=FauxmoESP-3.0.0
JOB 1627 - 2018/10/05 17:29:19 Error syncing library: Error while zipping library: Symlink not allowed: .../bitbucket.org/xoseperez/fauxmoesp/examples/lib/fauxmoESP -> ../../src
JOB 1627 - 2018/10/05 17:29:19 Checking out tag: 3.0.1
JOB 1627 - 2018/10/05 17:29:19 path=bitbucket.org/xoseperez, archivename=FauxmoESP-3.0.1
JOB 1627 - 2018/10/05 17:29:19 Error syncing library: Error while zipping library: Symlink not allowed: .../bitbucket.org/xoseperez/fauxmoesp/examples/lib/fauxmoESP -> ../../src
```
We don't accept libraries with symlinks or exacutable binaries since we cannot manually check if the behaviour is malicious, sorry

closed 2018-10-05 by facchinm

xoseperez, 2018-10-07:
Got it. I fixed it, will it be updated automatically?

per1234, 2018-10-07:
Even though the symlink has been removed from the repository, it still exists in the tagged versions >2.3.0:
https://bitbucket.org/xoseperez/fauxmoesp/src/9a03a6124f7d6324c256eba3104578dd7c3dde2a/examples/lib/fauxmoESP?at=dev&fileviewer=file-view-default
The Library Manager indexer will not pick up any tag that contains a symlink. Once a tag is created that does not have a symlink, it will be picked up automatically within ~1 hr. So you need to either create a new tag or else redo the existing tag(s).

xoseperez, 2018-10-08:
That makes sense. I have tagged a new version and it got updated. Thank you very much.