arduino/Arduino #8070: feature request: Serial.print() easy concatenation

robzach, 2018-10-08:
Instead of having to write `Serial.print("val = " + String(val))` it would be preferable for `Serial.print` to cast all arguments to strings, i.e. `Serial.print("val = " + val)`.

(Note that the latter *will* compile in 1.8.7, but will produce a torrent of garbage output when viewed with the serial monitor.)

per1234, 2018-10-09:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3546

closed 2018-10-09 by per1234

matthijskooijman, 2018-10-09:
Note that the request in this issue is different from the proposal in that issue (this requests addition of strings, that allows multiple arguments to print), but since the result is the same (and addition of strings will likely not be easy to implement), marking this as a duplicate makes sense.