arduino/Arduino #8086: 10CL016 or 10CL025

Arkylien, 2018-10-10:
Is the FPGA chip 10CL016 or 10CL025?
here at the tab "tech specs" it states 10CL016:
https://store.arduino.cc/arduino-vidor-4000
and in the "documentation" tab it states 10CL025:
https://content.arduino.cc/assets/vidor_c10_sch.zip

I’m confused. what’s what?

facchinm, 2018-10-11:
Hi @Arkylien ,
the FPGA is a 10CL016; since the chips are pin to pin compatible we probably added the wrong symbol to the schematics. @pnndra can we update the schematics with the right symbol?

Arkylien, 2018-10-11:
Hi,
Thanks for the info. I don’t really have a preference. But for the buyers it is more clear if the 16 version is used as schematic. However it is interesting for some to see the 25 version circuit, maybe it can be placed at another place. But I'm new here so my opinion is not well based