arduino/Arduino #8089: advertisement covering reference home page

jcanzler, 2018-10-12:
Your MKR family advertisement is making it difficult to use the reference page because it is block key portions of the text and syntax.

per1234, 2018-10-12:
Here's a screenshot of the problem:
[image]
It doesn't occur for me with Firefox, only the Chrome browser.

per1234, 2018-10-12:
The problem no longer occurs for me after a Ctrl+F5 refresh so that makes me suspect the problem was fixed, but that the website team didn't bother to close this issue report. @jcanzler does the problem still occur for you?

smellai, 2018-10-15:
Yes please try to clear browser cache and check if looks better

closed 2018-10-15 by matteosuppo