arduino/Arduino #8090: Comment/uncomment on non-US keyboards

teo1978, 2018-10-12:
I use a Spanish keyboard.

Next to the menu item Edit->Comment/Uncomment, the hint for the corresponding keyboard shortcut is
```
Ctrl+Slash
```

On the Spanish keyboard, there is no Slash key. The slash character is done with Shift+7.

I have tried both Ctrl+7 and Ctrl+Shift+7 and neither works.

So either the shortcut is not compatible with all non-US keyboards, or the shortcut is not correctly indicated in the menu.

matthijskooijman, 2018-10-12:
I believe this is essentialy the problem described at https://github.com/bobbylight/RSyntaxTextArea/issues/157 which is the component we use for the editor area.

dsyleixa, 2018-10-17:
same to me on German Keyboard

ceinem, 2022-08-05:
Is there any update on this?

per1234, 2022-08-05:
@ceinem please check to see whether the issue still occurs with the latest nightly build of Arduino IDE 2.x. The download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

Let us know what you discover.